cocotb/cocotb #502: Is Python 3.5 supported for example ?

MortenZdk, 2016-12-12:
On Windows 7 64-bit system, with MinGW and Python 3.5.2, and Python 3 in path, I try to run the example at http://cocotb.readthedocs.io/en/latest/quickstart.html, but when doing `make` in MSys terminal, it gives:

```
$ make
  File "<string>", line 1
    from distutils import sysconfig; print sysconfig.get_config_var("DESTSHARED"
)
                                                   ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    from distutils import sysconfig; print sysconfig.get_config_var("DESTSHARED"
)
                                                   ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
/home/mz/cocotb/makefiles/simulators/Makefile.icarus:49: *** "Unable to locate command >iverilog<".  Stop.

```
Looks like the scripts does not support Python 3 with print being a function.

Is Python 3.5 supported for example, and what should I then do to use Python 3 ?

Thanks.

jeremyherbert, 2016-12-12:
Hi @MortenZdk 

I certainly use cocotb with python 3.5 on Linux and OS X. The problem here is that the Windows makefile is a bit wonky. You can have a look at the Linux one to see how it should be working.

I can immediately see that these lines can be fixed by something like this:

`LOCAL_PYTHON_VERSION?=$(shell $(PYTHON_BIN) -c 'from __future__ import print_function; import distutils.sysconfig; print(distutils.sysconfig.get_python_version())')
PYTHON_DYNLIBDIR:=$(shell $(PYTHON_BIN) -c 'from __future__ import print_function; from distutils import sysconfig; print(sysconfig.get_config_var("DESTSHARED"))')`

Unfortunately I can't test this for you because I am not near my computer (I blame any typing errors on my tablet), and I also don't use Windows for this kind of work ;) but keep reporting back if you find issues; there is no simple way to do regression testing for Windows using free services like there is for Linux.

brunopen, 2016-12-12:
+1 for having it working with Python 3.5 with Debian. 
Agreed the problem should be Windows/Msys

MortenZdk, 2016-12-12:
Hi @jeremyherbert,

Thanks for looking into this, and nice to know that there should not be any major blocks for using Python 3.5.  I will try to update the scripts, and give a reply here.

Best Morten

javValverde, 2016-12-12:
@jeremyherbert I guess [appVeyor](https://www.appveyor.com/) could be used for continuous integration in Windows.

jeremyherbert, 2016-12-12:
@javValverde Yes, that is true, but to be honest I have no reason to do it myself. I think a much bigger issue for this project is the fact that basically all of the simulators used in industry are very proprietary and cost significant amounts of money. They can't really be used in CI for this project unless someone runs it locally, donates licenses, etc. So when someone makes patches to fix questa, it can break Riviera and no-one knows about it. 

I'm not even sure if the mega corporations in this space would be happy to let some random person on the internet submit code to their simulators via pull requests. but this is getting off topic (unless you, the person reading this, work at mentor/cadence/synopsys/aldec and can give out licenses! Get in touch please ;) )

ilesser, 2016-12-12:
Hi @jeremyherbert,

I am new to cocotb and I think it is great!
The only issue I am having is making it use Python 3 to run the test.
I've tried this "PYTHON_BIN=python3 PYTHON_PATH=/path/to/my/python/install make"

Is there anything else I should do?
I've also tried modifying the Makefile.pylib.Linux but without success

Thanks,
Ignacio

jeremyherbert, 2016-12-12:
@ilesser, are you on Windows? If not, could you please create a new issue for this?

ilesser, 2016-12-12:
I am in linux, creating new issue.

MortenZdk, 2016-12-12:
Appears that `print` for Python is used as a statement very few places, where I found:
- makefiles\Makefile.pylib.Msys
- bin\create_files.py
- cocotb\binary.py
- cocotb\drivers\ad9361.py
- tests\test_cases\test_avalon\test_avalon.py

Several places it was actually outcommented code.

To make code compatible with both Python 2.6+ and 3+, I updated the `print ...` to below:

    import sys
    ...
    sys.stdout.write('%s\n' % ...)

I would like to suggest that this is done in general for cocotb, but since I am pretty new to this (started today :-), I think that somebody more experiences should make this judgement.

jeremyherbert, 2016-12-13:
@MortenZdk do the print statements as-is cause the tests to fail on your PC?

MortenZdk, 2016-12-13:
@jeremyherbert Yes cocotb fails if used with Python 3, since Python is called in e.g. the makefiles/Makefile.pylib.Msys with the one-liner script `print distutils.sysconfig.get_python_version()`, and such use of `print` as a statement results in an error in Python 3.

jeremyherbert, 2016-12-13:
How about if you fix those two lines and none of the others?

MortenZdk, 2016-12-13:
When I fixed the 3 lines with `print` in the in makefiles/Makefile.pylib.Msys then the endian_swapper example ran OK using Python 3.5.  There are other lines with `print` statements in other .py files in the checkout, but these did not cause problems for the endian_swapper example.

jeremyherbert, 2016-12-13:
Where in the endian swapper example is the erroneous line? I can't seem to find it, and the endian_swapper test appears to work fine with python 3.4 on travis: https://travis-ci.org/potentialventures/cocotb/jobs/180455755

jeremyherbert, 2016-12-13:
Ah my apologies, I misunderstood your post. It's good that it is working for you. The only print statements are in docstrings; while that should be fixed, it's not critical. But if you'd like to submit a patch I'm sure it would be welcome.

MortenZdk, 2016-12-13:
Thanks for looking into this, and I will see if I can figure out to submit a patch... :-)

eric-wieser, 2019-01-13:
@imphil I think this is fixed too, searching for `print ` finds nothing relevant