cocotb/cocotb #503: How to use Python 3 in Red Hat?

ilesser, 2016-12-12:
I am working in a personal project in a university hosted server using Red Hat 4.4.
It has Python 2.6.6 installed because the repositories are old for RH4.4 .
I compiled and installed (locally) Python 3.4.5.

I am working with the example
I've tried to use my python 3 version to run the simulations but somehow I can't make it work.
This is the makefile I am using:
```
# Default to verilog
TOPLEVEL_LANG ?= verilog
# Set cocotb directory
COCOTB=$(HOME)/git/cocotb
# Set verilog sources
WPWD=$(shell sh -c 'pwd -W')
PWD=$(shell pwd)
VLOG_DIR = $(PWD)/../../../rtl/source
VERILOG_SOURCES = $(wildcard $(VLOG_DIR)/*.v)
# Define top level and test module
TOPLEVEL = bkm_steps
MODULE = test_rand
# Simulator dependant options
SIM = vcs
COMPILE_ARGS = +incdir+$(VLOG_DIR)
COMPILE_ARGS += -top bkm_steps
COMPILE_ARGS += +vcs+vcdpluson -debug_pp
# Include cocotb makefiles
include $(COCOTB)/makefiles/Makefile.inc
include $(COCOTB)/makefiles/Makefile.sim
```

There are the two things I've done, one of them is to set variables before calling make.
```
PYTHON_BIN=python3 PYTHON_PATH=/path/to/my/python/install make
```

The other one is modifying the Makefile.pylib.Linux to use python3
```
ifneq ($(COCOTB_PYTHON_DEBUG),)
DBG_PYTHON_VERSION=$(shell python -c 'from __future__ import print_function; import distutils.sysconfig; print(distutils.sysconfig.get_python_version())')
PYTHON_BIN=python3$(DBG_PYTHON_VERSION)-dbg
else
PYTHON_BIN=python3
endif
```

I my test_rand.py I just print the version from sys `print(sys.verion)` but keep getting the 2.6.6

Is there something else I should be doing? Am I missing something here?

Thanks,
Ignacio

themperek, 2016-12-12:
What is your error?

Try to use conda: http://conda.pydata.org/miniconda.html

ilesser, 2016-12-12:
Actually I am not getting any error.
I am trying to find a way to use my local python version (3.4.5) instead of the old default one (2.6.6).

I would like cocotb to simulate using Python 3 binary rather than Python 2.
But I don't know how to do it.

I'll try using conda as you suggested, but then the question would be "How do I make cocotb use conda?"

Thanks,
Ignacio

elgorwi, 2016-12-12:
I've used anaconda 2 before on a linux mint system together with modelsim. 
I had to set the PYTHONHOME variable to point to the 'anaconda2' directory to ensure that the anaconda python rather than the system python was picked up by cocotb.

jeremyherbert, 2016-12-13:
If you don't care about using the system python, you can just use

`PATH=/path/to/anaconda/bin make`

I just set this in my .bashrc as I don't care about using the system python (I'm using anaconda3 for all of my systems which run cocotb). You also need to add PYTHONHOME as well, see https://github.com/potentialventures/cocotb/issues/444

ilesser, 2016-12-15:
Setting the PYTHONHOME and PATH made the trick!
Thanks
Close #503

closed 2016-12-15 by ilesser