cocotb/cocotb #505: Return value from @cocotb.coroutine

LinuxFanatic777, 2016-12-13:
I have such Python 3 code to access registers of my DUT:
```
@cocotb.coroutine
def write_reg(dut, reg, value):
	dut.cc.brdg_addr = reg
	dut.cc.brdg_din = value
	dut.cc.brdg_wr_n = 0
	yield posedge(dut.clock100)
	dut.cc.brdg_wr_n = 1
	yield posedge(dut.clock100)

@cocotb.coroutine
def read_reg(dut, reg):
	dut.cc.brdg_addr = reg
	dut.cc.brdg_rd_n = 0
	yield posedge(dut.clock100)
	dut.cc.brdg_rd_n = 1
	yield posedge(dut.clock100)
	yield posedge(dut.clock100)
	res = int(dut.cc.brdg_dout)
	raise ReturnValue(res)
```
It's not a problem to call **write_reg**, **read_reg** also working but it's takes me three more lines and more variables:
```
		yield misc.write_reg(dut, 0x37, 123)

		gen__reg_0x37 = misc.read_reg(dut, 0x37)
		yield gen__reg_0x37
		reg_0x37 = int(gen__reg_0x37.retval)
		dut.log.info("TEST READ reg_0x37= %d" % (reg_0x37))
```
As result I see correct value in 123, but... I want to do this in **single line**, just like this:
```
reg_0x37 = misc.read_reg(dut, 0x37)
```
If my understanding of Python is correct I can do something like this:
```
def __read_reg(dut, reg):
	try:
		for value in read_reg(dut, reg):
			print("value= %s" % str(value))
	except ReturnValue, value:
		return int(value)
	return 0
```
read_reg creates generator here and iterate it up to it's end and catch ReturnValue with result of reading. Unfortunately most approaches I tried gives me **iter() returned non-iterator of type 'RunningCoroutine'** all the time =((( I spent huge amount of time to resolve this by learning Python language articles and it seems that this limitation is cocotb specific...

leftink, 2016-12-13:
@LinuxFanatic777 You almost had it.  To do the read all you need to do is:
`reg_0x37 = yield misc.read_reg(dut, 0x37)`

LinuxFanatic777, 2016-12-13:
Thank you very much!
```
@cocotb.coroutine
def read_reg(dut, reg):
	dut.cc.brdg_addr = reg
	dut.cc.brdg_rd_n = 0
	yield posedge(dut.clock100)
	dut.cc.brdg_rd_n = 1
	yield posedge(dut.clock100)
	yield posedge(dut.clock100)
	res = int(dut.cc.brdg_dout)
	raise ReturnValue(res)
...
		yield misc.write_reg(dut, 0x37, 123)
		gen__reg_0x37 = misc.read_reg(dut, 0x37)
		yield gen__reg_0x37
		reg_0x37 = int(gen__reg_0x37.retval)
		dut.log.info("TEST READ reg_0x37= %d" % (reg_0x37))

		yield misc.write_reg(dut, 0x37, 157)
		read = yield misc.read_reg(dut, 0x37)
		dut.log.info("NEW READ read= %d" % read)
```
This helps! I see correct 157 integer in **read** variable!

closed 2016-12-13 by LinuxFanatic777

eric-wieser, 2018-07-30:
Added documentation for this in #634