cocotb/cocotb #506: Unclear Error message

brunopen, 2016-12-13:
Hi all,
I am facing the following Error message:
Cocotb initialisation failed - exiting

This appears here in the log of my test:
Loading /home/bpe/digflow_repo/digital_flow/trunk/verif/cocotb/build/libs/x86_64/libvpi.so
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
Cocotb initialisation failed - exiting
   0.00ns INFO     cocotb.gpi                                  gpi_embed.c:248  in embed_sim_init                  Running on ModelSim for Questa-64 version 10.4c_5 2015.11
   0.00ns INFO     cocotb.gpi                                  gpi_embed.c:249  in embed_sim_init                  Python interpreter initialised and cocotb loaded!

This message is not really helpful, and appears when I use python3.4 (the test works with python2.7).
More precisely if cocotb was compiled with python 2.7 (all the so building in cocotb/build), it works, even if then i run my test with python2 or 3.
But if the build in cocotb/build was achieved with python3 (and goes fine with no error), then I get the "Cocotb initialisation failed - exiting", no matter if I run it with python2 or 3.

I am a little bit puzzled by what is creating the problem, and how to debug it.
Does someone have some good ideas to investigate?

Thanks for your help!

jeremyherbert, 2016-12-13:
Hi @brunopen 

I assume you are using a relatively standard linux distro. What is the output of the following commands?

`which python`
`which python-config`
`python -c "from __future__ import print_function; import sys; print(sys.version_info.major, sys.version_info.minor)"`
`python -c "from __future__ import print_function; from distutils import sysconfig; print(sysconfig.get_python_version(), sysconfig.get_config_var('LIBDIR'), sysconfig.get_config_var('DESTSHARED'))"`
`echo $PATH`
`echo $PYTHONPATH`
`echo $PYTHONHOME`

brunopen, 2016-12-13:
Hi, @jeremyherbert.
First thanks for your immediate answer!
I am running on a debian machine. 
Here is the result in the failing case (using 3.4)

>which python
/usr/bin/python

>ll /usr/bin/python
/usr/bin/python -> python3.4

>which python-config 
Returns nothing. Is it normal?

> python -c "from __future__ import print_function; import sys; print(sys.version_info.major, sys.version_info.minor)"
3 4

> python -c "from __future__ import print_function; from distutils import sysconfig; print(sysconfig.get_python_version(), sysconfig.get_config_var('LIBDIR'), sysconfig.get_config_var('DESTSHARED'))"
3.4 /usr/lib /usr/lib/python3.4/lib-dynload

> echo $PATH
/opt/questasim/bin/:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games

>echo $PYTHONPATH
Empty, but I set it to the root of my test when running the test

>echo $PYTHONHOME
Empty

Hope it helps...

jeremyherbert, 2016-12-13:
Are you using the built in Debian python 3.4? Did you symlink it yourself ?

How about 

`which python3-config`

brunopen, 2016-12-13:
Hi, 
I install the 3.4 from apt-get install (3.4.2) , installed python and python-dev
I linked the python to python3 by myself since python2.7 was installed before, and was therefore hte original default.

>which python3-config
/usr/bin/python3-config
>ll /usr/bin/python3-config
/usr/bin/python3-config -> python3.4-config

I include again my edit of the previous message you may have missed:
Please note that the example (at least the endian swapper) are not affected by the problem...
My main issue it that verif files are pretty different...

jeremyherbert, 2016-12-13:
Sorry, I'm not sure I understand. What are the "verif" files? Are you able to run the endian_swapper example? And are you able to run the adder example? Have you tried running using iverilog instead of questa?

If you compile cocotb for python2 it should crash when running with python3 and vice versa. This is because one of the changes between the python versions was the name of the entry point function for compiled extensions. Further detail here: https://docs.python.org/3/howto/cporting.html

You might also want to try symlinking python3-config to python-config

brunopen, 2016-12-13:
Sorry, my fault. 
Yes I am able to run endian_swapper example (and others, checked in-between).
What fails is the verification for a block I designed. 
My problem is that I have it working on the example (for both python versions), but only working if I build cocotb with python2, and the Error message is not helping me enough fro my level to understand what is going wrong.
 
I am in the meantime trying to reduces the differences with examples.

Ok, what you say about the uncompatibility between python version doesn't match with my understand of what is going on... I must be wrong somewhere...

brunopen, 2016-12-13:
Ok, I think I have isolated pretty well the problem.
The problem happens in my case when:
-I am using python3 (as explained before)
-A Makefile invokes the usual simulation Makefile in this fashion:
run_sim:
	@$(MAKE) -C cocotb/examples/dff/tests/

In this case, using this Makefile leads to Cocotb initialisation failed - exiting

Changing the Makefile to
run_sim:
	cd cocotb/examples/dff/tests/ && $(MAKE)
works.

Make version in use: 
/usr/bin/make --version
GNU Make 4.0

jeremyherbert, 2016-12-14:
What about if you run it via  `strace`? Do you get any useful errors which show up? Or funny signals?

Also, perhaps you can try running `nm -D` to list the symbols in the compiled simulator library. If it has a function called `PyInit_simulator` it is being compiled for python3, otherwise if it has `initsimulator` then it is being compiled for python2. If it has neither of those, something is badly wrong ;)

Is the code to your block and makefile something you'd be willing to share?

brunopen, 2016-12-14:
Hi @jeremyherbert 
Thanks for your suggestions. 

strace is not showing anything strange, and I haven't seen any (relevant) difference between a phyton 2 and phyton3 run in the trace.

I am not clear how to use your "nm -D" advice. Where should I add this?

I am working with VHDL, so I cannot use Iverilog. I will give a try with GHDL, good point.

For the confidentiality of my code, I need to check ;-)

jeremyherbert, 2016-12-14:
Something like this is how you use nm: http://stackoverflow.com/questions/4514745/how-do-i-view-the-list-of-functions-a-linux-shared-library-is-exporting

brunopen, 2016-12-14:
Thanks, I will look at it. Also useful for my general knowledge.

With successive iterations, I think I know have isolated the problem. It seems to be related to  the way I import other modules in the testbench.
I include my dut model that is in a different folder (../model from testbench py file), and common functions that I want to share among all my blocks.
I've seen that import has change between python 2 and 3, so that could be the explanation...
I have to find the proper way to do it in python3.

jeremyherbert, 2016-12-14:
Ok, sounds good. This might help: http://stackoverflow.com/questions/448271/what-is-init-py-for

brunopen, 2016-12-14:
Hi @jeremyherbert.

It seems that the include is not the only problem...
And from what I see, the main issue I face is, I realize now, back on the original topic, that with python3, I have errors with no trace, that are a pain to investigate.

I looked in simulator.so in build folder in cocotb, and I find indeed 
when compiled with python3, I find 0000000000003f46 T PyInit_simulator and I can't find initsimulator 
when compiled with python3, I find 0000000000003dde T initsimulator and I can't find PyInit_simulator

It seems in line with what you mentionned...

brunopen, 2016-12-14:
@jeremyherbert : I have developed a simple testcase. I am not sure it illustrates the real problem I am facing, but it illustrates the lack of message in addition with Cocotb initialization failed - exiting.

What would be the easiest way to share it with you?

jeremyherbert, 2016-12-15:
Ok it sounds like you're making progress. Does your test work after you use the correct name?

jeremyherbert, 2016-12-15:
If you'd like to share the exact code in your simple testcase here, I can try to run it on my system.

brunopen, 2016-12-15:
Hi @jeremyherbert 
This is still not working on my side, this lack of visibility of the problem is driving me nuts.
Thanks again for your support.

I have packaged my testcase in the enclosed tar archive. Please note that tar is not a supported attachment format, therefore I renamed it as txt...
Once untared, 
You can run it from the Makefile at top with   
make sim block="fpga/testcase" SIM=questa
Note that I am working with Questa, 64 bits

This design should fail since no model is described/imported, but a model is used.
This is properly reported with a traceback in python2 (starting with "Send raised exception: 'Testbench' object has no attribute 'model'")  , but just with "Cocotb initialization failed - exiting" in python3.

I hope this testcase is clear enough and minimal enough...

I am curious to know how it works on your side (in python3, since python outputs what I expect...)

[testcase.txt](https://github.com/potentialventures/cocotb/files/654231/testcase.txt)

brunopen, 2016-12-15:
Some other interesting pieces of information:
-The problem can be reproduced in an even easier fashion by commenting the model definition of hte cocotb of an example, dff for instance.

-When doing the same test with icarus (instead of questa), I get the track even in python3...

brunopen, 2016-12-21:
Hi @jeremyherbert , do you have any feedback about this ?

jeremyherbert, 2016-12-21:
Sorry for the delay on this, life got mega busy all of a sudden. I'll try to look at this tonight.

brunopen, 2016-12-21:
No problem, no stress, take the time you need, your support is already very much appreciated ;-)

themperek, 2019-10-16:
It should not be a problem anymore. In case you still have issues please reopen. 

@cmarqu  thank you for pointing out.

closed 2019-10-16 by themperek