cocotb/cocotb #507: Issue with input larger than 64 bits

brunopen, 2016-12-21:
Hi all,
I have in my block an input that is 78 bits. 
When I drive it to a value say "n", what I see is that the 64 low bits are driven with n , and the top 14 bits are driven with the lsb, as if the driving value would be repeated every 64 bits.

If I drive "10" in decimal on my bus, 
I expect to see : h000000000000000A
I see h000A00000000000A

I see the same h000A00000000000A driven in the model and the simulation.

I tried by explicitly casting the signal driving the value to a long, but it doesn't help.
Any idea? 

Thanks

brunopen, 2016-12-21:
To complete my previous post, running cocotb in debug, I see the following when driving 100 (decimal)
FliObjHdl.cpp:307  in set_signal_value                set_signal_value(long)::0x0000000000000064
FliObjHdl.cpp:291  in get_signal_value_binstr         Retrieved "000000011001000000000000000000000000000000000000000000000000000000000001100100" for value object sample_DI

leftink, 2016-12-21:
Try driving the signal using BinaryValue instead of with an int and see if that makes a difference.

brunopen, 2016-12-21:
Hi @leftink,
Thanks for your suggestion.
I defined my signal as sample = BinaryValue(value=100,bits=78)
This leads to exactly the same result as before, the "100" is duplicated on bits 64 and more

I also checked the behavior of sample = BinaryValue(value=100,bits=78,bigEndian=True )
In that case, it drives properly the bus, driving:
110010000000000000000000000000000000000000000000000000000000000000000000000000

So the problem appears to be with binary value if the size is larger than 64 bits, but none of the bits beyond the 64 first bits is set to 1.

brunopen, 2016-12-21:
I tried by directly giving a bistring, like :
sample = BinaryValue(value="000000000000000000000000000000000000000000000000000000000000000000000000000001",bits=78,bigEndian=False)
and it fails the same, I end up with 000000000000010000000000000000000000000000000000000000000000000000000000000001 driven by the driver

brunopen, 2016-12-21:
Hum... I found a case where it seems to work with the binary value. I'll confirm as soon as it gets clearer. 
But the issue is there for sure when not using BinaryValue.

brunopen, 2016-12-21:
It seems that the problem happens when only a small value is loaded in Binary value.
For instance BinaryValue(value="000000000000000000000000000000000000000000000000000000000000000000000000000001",bits=78,bigEndian=False)
is driven wrong, but
BinaryValue(value="000000000000000000000000000000000000000000000100000000000000000010000000000001",bits=78,bigEndian=False)
is driven with hte correct value...

leftink, 2016-12-21:
I am not sure what the deal is with BinaryValue not always working, do you code you could share?

Driving with an int will never work and should probably throw an error.  The reason is the int can only provide up to 64-bits so there isnt enough info to drive a larger signal.

brunopen, 2016-12-21:
Hi @leftink. 
Thanks for the proposal. I will consider to export a testcase out of it if I can't get any progress. Unfortunately, the hdl code is confidential.
My guess at the moment is that depending on the size of the signal I drive in python, it is mapped on a different type of variable in C++, and the "expend" to more than 64 bits of something coded on less than 32 bits (ie a python int) goes wrong...

leftink, 2016-12-21:
You are correct, but the buffer should be fully populated when using a BinaryValue which is why I was surprised you had issues with that.

brunopen, 2016-12-21:
What I have found is that the change of behavior corresponds to the change of set_signal_value call: set_signal_value(long)::0x000000003fffffff (buggy)
set_signal_value(string)::000000000000000000000000000000000000000000000001111111111111111111111111111111 (correct behavior).
What I am looking at, at the moment, is that in hte handle.py, there is the choice between using a string or a int (in setimmediatevalue in hanlde.py).

leftink, 2016-12-21:
Yes, the set_signal_value(long) is the interface I was saying is limited to only 64-bits.  It is a faster interface than driving by binary strings.

brunopen, 2016-12-21:
Yes, it makes sense.
The main problem is that in my case, if the content fits on 32 bits, it goes thought this function, and at the "extension" of the signal to the 78 bits of my bus (or whatever further than 64 bits), something goes wrong and bits i+64 get driven by value on bit i instead of being driven to 0.

elgorwi, 2018-06-04:
@brunopen I've proposed a fix in #621.

themperek, 2019-10-16:
It should not be a problem anymore. In case you still have issues please reopen. 

@cmarqu  thank you for pointing out.

closed 2019-10-16 by themperek