cocotb/cocotb #508: Ability to parallelize regression

brunopen, 2017-01-19:
Hi all,
I created a regression made of several factories, all using the same run coroutine.
The regression generated is run sequentially.
I would be interested in a way to parallelize the execution of the tests inside the regression to reduce the simulation time. 
Did I missed the feature implementation?
What would be the easiest way to achieve it if it is not implemented in Cocotb, by dispatching several sub regressions from a Makefile for instance?

thanks in advance fro your help !

LinuxFanatic777, 2017-01-24:
Have you tried to wrap this by some script that will run multiple instances of your cocotb-based tests with different parameters? I'm also thinking about this way to run many parallel tests on heavily-multicore server...

brunopen, 2017-01-25:
Indeed, that was my idea, but that would be easier (I am lazy) if directly handled by Cocotb.
If I generate several sub-regressions with a Makefile for instance, I'll need to run them in different folders (as far as I understand).
Once all the sub-regressions are complete I would need to merge the results and the coverage for instance.

Do you see a simpler way to achieve this?

LinuxFanatic777, 2017-01-28:
It would be great to see this mode as built-in function in Cocotb. Did you noticed that tempo of commits to this project (I mean Cocotb) is slowed down? I think developers created all things they want and now it's time for us (users) to make some involvement... ;)
If I will work in verification for longer time then I could be comitter of this project.

chiggs, 2017-01-28:
@LinuxFanatic777 just wanted to provide an update... you're right things have slowed down a bit, this is down to a change of circumstances about 7 months ago.  Apologies for the lack of responses things have been pretty busy.  However we do have some changes to push upstream and hope to engage more with the project during 2017!

Regarding this issue - we have historically dispatched multiple invocations outside of Cocotb and then combined the results together.  There is a helper script here for combining results together: https://github.com/potentialventures/cocotb/blob/master/bin/combine_results.py

I agree it would be nice if this was neater and better integrated with Cocotb itself.  We should think about how to improve the building and running of tests as it can get a bit clunky at the moment.

brunopen, 2017-01-28:
@chiggs, thanks for your reply and for sharing the script. 
@linuxfanatic777, @chiggs, we all agree about the value of such a feature.

jahagirdar, 2019-03-16:
How do we control the name/location of the results.xml file for each parallel run?

in my testbench I have `$readmemh(romfile, DUT.imem.memory);`
Where romfile is unique to each testcase and is loaded at runtime via a `plusarg`. I am not sure whether this can be done inside the cocotb framework. So I have separate make targets for each test  `make sim SIM_ARGS=+romfile=testcases/foo.rom`

How do I ensure that the result of each target goes to a separate xml file? e.g. the above goes to `foo.xml` or `foo/results.xml`

jahagirdar, 2019-07-23:
XUnitReporter class has an option of `filename="results.xml"` This option is not used in regression.py where this class is used.
I have locally tested the following change.

```
 if os.getenv('RESULT_PATH'):
              self.xunit = XUnitReporter(filename=os.path.join(os.getenv('RESULT_PATH'),"results.xml"))
          else:
              self.xunit = XUnitReporter()
```
@cmarqu do you see any problems with this? Should I submit a pull request?

ktbarrett, 2024-02-28:
The `COCOTB_RESULTS_FILE` variable can be used to point the results.xml to different locations so parallel runs don't conflict. #3732 would include the `combine_results.py` script into the package, so the reports can be joined. But automating the discovery and parallel execution of tests is not something we plan on supporting in cocotb as it's out-of-scope.

closed 2024-02-28 by ktbarrett