cocotb/cocotb #509: ERROR at the end of regression with Questa

brunopen, 2017-01-26:
Hi all,
I am using Cocotb with Questa, and if I don't set the GUI=1, at the shutdown of the simulator, at the end of the regression, I get the following:

# 312537.50ns INFO     cocotb.regression                         regression.py:184  in tear_down                       Shutting down...
# 312537.50ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:128  in run_callback                    Generic run_callback done
# 312537.50ns DEBUG    cocotb.gpi                                 FliCbHdl.cpp:85   in cleanup_callback                Removing Postponed DELETE timer 0x1
# Simulation halt requested by foreign interface.
# 312537.50ns ERROR    cocotb.scheduler                            __init__.py:172  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
# 312537.50ns DEBUG    cocotb.scheduler                           scheduler.py:535  in finish_scheduler                Issue sim closedown result to regresssion object
# End time: 08:32:36 on Jan 26,2017, Elapsed time: 0:01:12
# Errors: 0, Warnings: 3

By rewinding the source of this message, my understanding is that it is triggered by FliShutdownCbHdl::run_callback , that triggers gpi_embed_end(), that generates embed_sim_event(SIM_FAIL, "Simulator shutdown prematurely");
(I am using VHDL, that's why I looked at the FLI).

The way I understand this shutdown is not "premature", and is triggered by the need to close the simulator at the end of hte regression. Is my understanding correct? Am I the only one getting this ERROR message?

Thanks for your help!

LinuxFanatic777, 2017-01-28:
Which operating system you are using? I'm not getting such error when GUI=0 both on Linux and Windows 7. Everything is fine, however my language is Verilog.

brunopen, 2017-01-28:
I have this with Debian. Unfortunately, I won't be able to test with another OS because of questa license. But thanks for your answer!

LinuxFanatic777, 2017-02-01:
You could try Modelsim Altera Starter Edition which is free for any OS. Also, try simple test with Verilog (not VHDL):
```
`timescale 1ns / 1ns

module top_cocotb(
input clock,
input reset,
output reg [3:0] test
);

initial test = 0;
always @(posedge clock) begin
	if(reset) test <= 0;
	else test <= test + 1;
end
endmodule
```

brunopen, 2017-02-01:
Thanks for the advice. 
I am currently away, I will come back in 3 weeks. As soon as I am back, I'll try this.

brunopen, 2017-05-10:
Hi guys, I fixed this, that was my fault. 
For info, I had a mistake in my makefile , having a VSIM_ARGS appearing twice, leading to a vsim command containing  "vsim -coverage -onfinish exit -foreign \"cocotb_init libfli.so\" -foreign \"cocotb_init libfli.so\" work.fs_fpga_top" >> obj_nosvn/runsim.do which was crating the issue reported.

I close it.

closed 2017-05-10 by brunopen