cocotb/cocotb #513: Is Cocotb dead?

Paebbels, 2017-02-16:
Hello,

Circa 6 month ago I wrote such a question in a polite way as an email directly to the authors. They responded and promised to be more active in Cocotb again.

We have seen circa 5 merged pull requests since then, while the number of open issues (including found bugs) and open pull requests (including new features and bug fixes) are rising. We can also notice that the number of user activity decreases.

I invested a lot of time to get Cocotb and GHDL with LLVM backend compiling on Windows. But should I share my patches with a seemingly dead project?

So for our project [PoC](HTTPS://Github.com/VLSI-EDA/PoC), I decided to stop investments into supporting Cocotb testbenches in parallel to normal testbenches for our shipped IP cores, unless Cocotb gets some significant improvements.

jeremyherbert, 2017-02-16:
Perhaps this could also be remedied by adding some new committers to the project?

Paebbels, 2017-02-16:
@leftink, @elgorwi, @mciepluc, @Martoni, @metaspace seem to be or at least were active on Cocotb. If needed I can provide a neutral ground for a repo were we can merge bugfixes and additions. From there we can still provide upstream fixes to this repo if the development should continue in the future.

jeremyherbert, 2017-02-17:
@Paebbels I'm happy to join in as well

metaspace, 2017-02-17:
We are still actively using cocotb, but my patches that allow cocotb to run with 32 bit python3 and Aldec Riviera on a 64 bit platform were rejected. I don't have the time to refactor them at the moment, hence the fork. #443

stuarthodgson, 2017-02-17:
Hi all, 

First cocotb is not dead. But we do have problems,  primarily that we have been too quiet and unresponsive to the community. We know this needs to change,  to this end we will update here with more detail on how we would like to move forwards over the weekend.

leftink, 2017-02-17:
@Paebbels, I am still developing, but work has taken me away for a bit.  I have a few bugs that I need to get fixed for our company to really use this.  Furtther, I am planning on developing a way where we can register data types to return other than Binary Value.   BinaryValue has several issues that makes it extremely hard to work with.

nevsan, 2017-02-17:
I've been leveraging this for awhile and haven't needed to make any patches. I would love to know that things are still active and may contribute some of the scripts that have helped me get going.

stuarthodgson, 2017-02-19:
The short answer is that no, Cocotb isn't dead - far from it!

However it is clear that we need to radically change how things work since at the moment other demands on our time are obviously holding the project back.

This load comes from two areas; just being busy, and the breadth of the project as it now stands. Some of the features now present are not ones we use such as GHDL support, Windows etc.  All are areas that we know have value to the community but not where we have any experience or even ability to test. How do we get a regression test running on windows for instance?

To solve this, we'd like to propose a more community-driven development process that will hand over some control of the project (and with it the associated workload) to a set of maintainers.  The maintainters will be comprised of volunteers who have contributed towards Cocotb and are willing to take on responsibility for a particular area they are familiar with.

To transition Cocotb to a more mature community-driven project, we think it's necessary to more formally define a development model and contribution process.  In particular for new features or API additions or changes it's important to achieve consensus and provide visibility to existing users ahead of time.  This will mirror the Python model where proposals are debated via PEPs before they are implemented, although we will want something more lightweight.

The goal will be to achieve a consensus from developers and users, with maintainers and the project founders on hand to resolve any disputes or assist in deciding between competing options.  Like Python, we will continue as benevelent dictators or arbiters of last resort if no agreement can be reached.

In the short term we'll need to do the following:

* Appoint a set of maintainers who can be responsible for reviwing and approving code
* Create a contributing guide which defines the development process and provides an objective criteria that reviews can apply to PRs.

## Contribution Process

Our inital thoughts for the contributing process are as follows:

### Bug Fixes

1. Raise an issue describing the problem
2. If applicable, create a unit test to cover the problem
3. Any commits addressing the problem should reference the issue
4. Raise a PR
5. PR is reviewed by maintainers. Assuming all regressions pass and backwards compatibility of APIs is maintained this review can be quite lightweight.
6. If maintainers approve the PR it can be merged

### New features

1. Raise an issue proposing a change or features. This should include any APIs and a use case example.
2. We should then as a community discuss issues such as API, performance implications, related features and reach a consensus.
3. A PR should be raised implementing the agreed feature, conforming to the contributing guidelines (coding style, unit tests).
4. The PR is reviewed by maintainers.
5. If maintainers approve the PR it can be merged

We hope that this will allow the review to more complete since reviewers will know more about the context and history of the issue. The aim is not to restrict development but to increase the chances that contributions will make it in from the outset by getting agreement up-front about the change, and provide increased visibility and engagement from users who can provide feedback at the definition stage rather than during PR review.

We will start adding the above as a contributing guide to the repository (which others can help define).

## Future focus

We'll also need to address some parts of the project that are currently falling behind the standards required of a larger group project.

* Release process, packaging and project versioning
* Documentation, examples, tutorials
* Quality of unit and regression testing
* Separation of core functionality from add-ons and extensions

Thanks to everyone who has got in touch to express their support for the project.  It's great to hear all the different ways that Cocotb is being used in the field.  Thanks also to @paebbels for raising this overdue discussion to the fore (and also others like @luked80  via e-mail).

Paebbels, 2017-02-19:
@stuarthodgson Thanks your reply and afford in opening up Cocotb for a more community driven approach.

**I would like to signup as a maintainer in the areas:**
* **Windows support** in general with special focus on
  * Python 3 (64 bit)
  * PowerShell scripting
* **GHDL support**
  * Linux
  * Windows
* **AppVeyor** (Windows based CI)

If needed I can also contribute in the areas:
* ModelSim/QuestaSim
* Travis-CI

For me and projects which I maintain or contribute to, it's important to be platform independent, so a Linux only project is a no-go for us.

luked80, 2017-02-20:
Thanks @stuarthodgson and @chiggs. 
This is great stuff to hear.
cocotb usage continues to grow inside Broadcom, and I'm forever thankful for your contributions.
I use cocotb daily and have a bunch of improvements and bug fixes that others might find useful.
Looking forward to the continued success of cocotb.

jeremyherbert, 2017-02-20:
I'm happy to help out where I can. I build Fpga hardware to be used in my research and cocotb makes it a lot easier to test it. I also do all of my signal processing in python, so testing is often as simple as importing that same code into a cocotb test.  

Things I am interested in:
- testing of cocotb itself
- improving error messages/crash messages
- OS X travis support
- Vivado simulation support
- waf/scons support
- cookie cutter templates for brand new IPs/ autodetection of files in simple projects
- cython integration
- continuing python3+ support
- documentation

I must admit that I have been holding off on patches because of the slow merge rate recently. But I think we all understand how these projects can become difficult to fit in time-wise.

Paebbels, 2017-02-20:
@jeremyherbert Vivado support? Does Vivado offer any API for external code? And if so, do you trust them?

They are still struggling to implement protected types (VHDL-2002) and other basic VHDL-2008 features.

jeremyherbert, 2017-02-20:
I am not a fan of vivado, trust me ;) especially after Xilinx dumped the 6 series chips. But I am unfortunately locked in to using it for other reasons. According to Xilinx they support systemverilog DPI, but I haven't used it myself. 

I unfortunately don't have access to another simulator which can read the Xilinx encrypted IPs and run on Linux. Otherwise I would be happy with iverilog.

Paebbels, 2017-02-20:
@jeremyherbert DPI (Direct Programming Interface) is from simulation code to the outside world.

Cocotb needs the other way around: **VPI** (Verilog Programming Interface) or **VHPI** (VHDL Programming Interface).

jeremyherbert, 2017-02-20:
Ah, my mistake; I can't say I ever looked into it in great detail so scratch that one. I guess I'll just go back to dreaming about a modelsim license. But the other features would still be useful I think.

leftink, 2017-02-20:
@jeremyherbert you mentioned WAF or SCONS.  I was originally a WAF supporter.  Our company is in the process rebuilding our build system and evaluated sevarl options.  We settled on DoIt.   It is a python based task automation that seems to follow KISS pretty well.

leftink, 2017-02-21:
@stuarthodgson and @chiggs I would like to help out where I can and I wouldn't mind becoming a maintainer.

mciepluc, 2017-02-21:
@stuarthodgson <https://github.com/stuarthodgson> and @chiggs
<https://github.com/chiggs>

Guys, I am happy to help as a maintainer in are of verification
flow/methodology (effectivelly - Python front-end only). This definetely
needs some clarificiations and comprehensive documentation to be prepared.
I find Cocotb extremly useful as a tool for effective advanced verification
bring-up. I invested a lot of time to understand the concept and it paid
off. However, some feedbeck I got from other verification engineers is that
actually it is difficult to understand the flow and examples are not
sufficient.

The merge request I created introduces some features known from
SystemVerilog, concerning metric driven verification. I was also happy to
see some discussions about  regressionss. My point is that it would really
cool to look at the Cocotb as for the mature verification flow. Accellera
is progressing slowly and Python-based verification could become a solid
alternative. I am going to discuss this at the DvCon next week.

On the other hand, if you disagree that methodological aspects of the
verification should become a part of the Cocotb, I am happy to create a
"Cocotb+" and leave the basic framework as it is, as a lower layer for new
extensions. The decision is up to you. Both approaches make sense.

wallento, 2017-03-03:
Hi all,

we currently heavily work on our CI for librecores.org and cocotb will be one of the promoted verification flows. In case you need some help for the basic maintaining tasks I would be happy to help, which will be documentation, examples and other bits and pieces. For example I currently struggle a lot with a centralized installation of cocotb, which is probably not the intended use case.

Cheers,
Stefan

themperek, 2017-03-04:
Hi All,

I am also willing to help.
Maybe we could have a mailing list or gitter/slack  chat channel to discuss this?

Cheers,
Tomasz

josko7452, 2017-03-17:
Hi All,

I was developing similar system to Cocotb using **MyHDL** cosimulation and **VUnit** for simulator invocation and result collection.
As I see it now, cocotb is far more mature in terms of co-simulation capability, so to not reinvent a wheel I would continue my efforts in Cocotb.

What I would like to implement is some degree of **VUnit** integration.
I was using **VUnit** before and I see that the invocation of the simulators and dependency resolution is far more mature in **VUnit**.

I.e. replace _make_ based ivocation system with _VUnit_ one.
Would there be interest for such an effort to include it in _Cocotb_?

Cheers
Bruno

mciepluc, 2017-03-25:
@stuarthodgson and @chiggs 
Any update, please?

@josko7452 
That sounds really good for me.

cmarqu, 2017-04-07:
I could act as regression tester for Cadence Incisive and Synopsys VCS-MX on Linux.
I'm also very pro a Slack/gitter channel.

LinuxFanatic777, 2017-04-19:
Don't know if there any critical bugs, however I think cocotb has absolutely complete feature set for my testing tasks. Can't imagine what else more I should dream about. I was able even run cocotb on Windows )))
In my company we planned to write some article at russian-speaking IT-blog and I suggested to write about "FPGA verification in Python language with cocotb". Of course it's important for me to see cocotb alive. Also I had finished my crossplatform python-based script which is starting same testbenches both on Linux/Windows and which is being configured by config-file. Due to high memory consumption by large projects I have idea to run tests step by step, some of them in parallel.
I can slightly participate in development of cocotb in designated areas. Python language is more preferable than makefiles - any plans to switch base?

LinuxFanatic777, 2017-04-19:
Also, interesting question, is it possible in the future to certificate cocotb to be used with DO-254?

Paebbels, 2017-04-19:
@LinuxFanatic777 How do you compile Cocotb on Windows? Makefiles are not supported on Windows.

For your question about missing features and issues:
* coverage data in general
* functional coverage
* Cocotb does not support "hot simulator switching" - which is a demand for our VHDL library [PoC](https://www.github.com/VLSI-EDA/PoC)
* some simulators are not supported on Linux or Windows
* Cocotb is based on makefiles introducing a lot of big dependencies; for instance a 2 GB MinGW setup
* The makefiles have missing dependencies causing memory side effects with outdated files.
* Cocotb is linking the SO or DLL files against the wrong libraries using the wrong compiler and with the wrong linker...
* Another issue is the linking of the VPI interface on Windows. It results in unresolved linker symbols / incomplete linking
* Only a limited set of simulator options is passed to the "user frontend".

As I wrote before, I have solution for some of the problems. But Cocotb is not active any more ...

Paebbels, 2017-04-19:
@stuarthodgson 2 month passed by....
Where are the life signs?

LinuxFanatic777, 2017-04-19:
> How do you compile Cocotb on Windows? Makefiles are not supported on Windows

https://github.com/potentialventures/cocotb/issues/498

stuarthodgson, 2017-04-19:
A small update. The folks for librecores have helped us set up a mailing list at ```https://lists.librecores.org/listinfo/cocotb```. In the coming weeks (I'm on holiday until may) we will be adding details of a maintainer list and contacting those that have expressed interest in contributing to this process. This will enable us to get though the list of PRs, review the outstanding issues as they are and get the years worth of commits we have upstream. 

Then we can start the discussions on new features and direction. I'll repeat what I've said already, cocotb is not dead, it's also not our day job though.

closed 2017-04-19 by stuarthodgson

themperek, 2017-11-04:
@stuarthodgson Can we get any update on this?