cocotb/cocotb #518: Question about bus driver at clock RisingEdge for back annotated simulation

brunopen, 2017-03-21:
Hi all,
This post is not to describe a bug, but to ask the community about the most elegant way of avoiding the behaviour that I am facing.
I am currently using Cocotb for gate level simulation, with back annotation.
My bus driver derives from the default cocotb bus driver, in sync mode.
In sync mode the BusDriver send operation happens at clock rising edge.
This makes no problem with RTL and 0 delay gate level simulation.

But in back-annotated simulation, I have a clock insertion delay of the same order as the input path, leading to setup violation on the input stage of flipflops (I have a race between data and clock, since data is launched on clock rise edge, which is the sensitive edge of my flipflops).

At the moment, I was considering launching my input data on the clock negedge, giving me 50% of a clock period has margin.

But I was wandering if using the driver send not in a synchronous mode would be better.

Please also note that I would be in favour of a solution that works in the same fashion for 0 delay and back annotated.

I hope my situation stating is clear, and any kind of input about it is welcome.

Best regards,

Bruno

luked80, 2017-05-02:
There's a couple of things you could do.

1. Overload drive() in Bus
You could overload the drive function within the Bus class and somehow include some propagation delay information.
Maybe create an additional _signals_delay dict within the Bus class instance.
And if there's a matching name entry assign the signal value after some delay.
For rtl or zero-delay, just don't populate the _signals_delay dict

2. Adding dut harness with some timing delays
Create a verilog test harness around your dut and make that the top level in your cocotb simulation.
And add some delay information to the signals that drive your dut in verilog.
You could use plusargs or similar to switch between rtl/zero-delay and sdf modes from the command line.

brunopen, 2017-05-05:
Very good proposals ! 

I like very much the 1), allowing no change in RTL with a default delay of 0, and allowing an potentially non 0 value of the GLS IOs.

Thanks !

closed 2017-05-05 by brunopen