cocotb/cocotb #520: Conflict between RisingEdge and ClockCycles

putsplut, 2017-03-25:
Using GHDL - don't know if this is part of the problem.

If I have both a yield RisingEdge(dut.clk) and a yield ClockCycles(dut.clk, num_cycles) active at the same time, say each in a forked coroutine, then there is an error message generated by arm_callback indicating "Attempt to prime an already primed trigger for cbValueChange!" .

The issue seems to be that both triggers are independent classes, and there is no check to see if a callback of RisingEdge is already active when a callback of ClockCycles is being requested.

A workaround is to make my own ClockCycles coroutine which uses RisingEdge N times.

imphil, 2017-04-28:
I get the same error with VCS, so it's not a GHDL problem.

luked80, 2017-05-02:
I don't see ClockCycles used in any of the included testcases.
Since the same thing can be achieved with a ClockCycles coroutine, I'm thinking we should just remove the ClockCycles trigger.

Is there any benefit to making Clockcycles a dedicated cocotb trigger rather than a coroutine?

putsplut, 2017-05-05:
I am in agreement.  If the callback only occurred after num_cycles, then there would be an efficiency gain.  As it is, the _check routine is called back to num_cycles times so I would not expect any significant performance advantage over a coroutine.

⁣Sent from BlueMail ​

leftink, 2017-05-05:
One reason to keep ClockCycles as a trigger is so it could be yielded back in a list of other triggers.  An example would be to use it as say a timeout if you expect a response in a certain number of clock cycles.

putsplut, 2017-05-05:
If ClockCycles is implemented as a coroutine, can’t you place the forked coroutine into the list of other triggers and accomplish the same thing?
I know I have done something like this, but I’m not in front of the correct computer at the moment.

luked80, 2017-05-05:
@leftink @putsplut you can add coroutines into a list that can be yielded using .join() e.g
```
forked_coro = fork(my_coro())
result = yield [Timer(10), forked_coro.join()]
```

putsplut, 2017-05-05:
The coroutine version is simple:
@cocotb.coroutine
def myClockCycles(signal, n):
    for _ in xrange(n):
        yield RisingEdge(signal)

This is inferior to the trigger code in that I only wrote it for rising edge.  But, the trigger version improperly checks to see if the signal is a “1” which results in the falling edge mode failing anyways.

In order to remain as a trigger, I think changes to the scheduler are needed in order to avoid the current issue of multiple value change callbacks being scheduled with the simulator.  There is a comment in the scheduler of:

"GPI triggers should only ever be pending if there is an associated coroutine waiting on that trigger, otherwise it would have been unprimed already.”

Since the trigger is not a coroutine, we either end up with no coroutines pending on the ClockCycles trigger, or we have an already primed trigger and the ClockCycles trigger will make the redundant call to register_value_change_callback.

To satisfy leftink's goal, I can have (per Luke’s comment):

proc = fork(my_process())
timeout = fork(myClockCycles(clk, 100))
result = yield [proc.join(), timeout.join()]

which is only slightly wordier than:
proc = fork(my_process())
result = wild [proc.join(), ClockCycles(clk, 100)]

putsplut, 2017-05-05:
I responded too quickly. The yielding to the trigger is obviously done from within a coroutine.

eric-wieser, 2018-08-09:
> One reason to keep ClockCycles as a trigger is so it could be yielded back in a list of other triggers

I think this would be achievable with a subclass of RunningCoroutine that sets `retval = self`, rather than needing to rework triggers

eric-wieser, 2019-03-21:
This is now fixed in master

closed 2019-03-21 by imphil