cocotb/cocotb #522: What is he most native way to use Cocotb for post synthesis simulation?

OrionBytes, 2017-04-07:
F.e. with Quartus

luked80, 2017-04-11:
I guess it depends on what your cocotb test environment looks like.

Do you have monitors probing into the hierarchy that your test relies on?
Attaching these monitors to a post synth netlist can be tricky (but not impossible).

The simplest approach is to have a test environment that only relies on access to top level inputs and outputs on your dut.
That way, you can just drop the post synth netlist into your rtl testbench and things should just work.

Can you share some more details on what your cocotb test environment looks like?

OrionBytes, 2017-04-12:
Thank you for the answer!
I will develop post-synthesis environment as a next step in project

luked80, 2017-05-02:
@OrionBytes How'd you go with this?

kirknodeng, 2017-06-13:
In my case, 
I defined some corner case variable in Makefile. 
and, when I run "make" at that time selecting corners. 
Like below:
ifeq ($C,WCL)
    CORNER = ssgnp0p72vm40c
endif
ifeq ($C,ML)
    CORNER = ffgnp0p88v125c
endif
ifeq ($C,LT)
    CORNER = ffgnp0p88vm40c
endif
ifeq ($C,WC)
    CORNER = ssgnp0p72v125c
endif
ifeq ($C,TT)
    CORNER = tt0p8v25c
endif

post netlist and DK is locased in VERILOG_SOURCES. 
and, I added one for sdf annotation. (sdf_annotation_$(CORNER).v)

Inside of "sdf_annotate_$(CORNER).v" have each corner case sdf annotation module.

module sdf_annotate();
initial begin
    $sdf_annotate("/my/sdf/sdf_file",dut,,"../sdf_annotate_ffgnp0p88v125c.log","MINIMUM");
end
endmodule

To avoid "x" propagation for gate-level simulation. (cdc path, metastable can generate 'x')
I used EXTRA_ARGS += +optionconfgfile+/my/timng/check/file.cfg

instance {my.cdc.metastable.path.reg} {noTiming};

this is my post simulation "VCS" example.

mciepluc, 2018-11-27:
@OrionBytes can this issue be closed? 
I think the above answers are clear. I am also developing sort of a "wrapper" myself around module that can be switched between RTL and gate-level versions. This is the easiest way.
The SDF annotation can be done also on the command line, but it actually depends on simulator.

ktbarrett, 2021-01-31:
Duplicate #1771.

closed 2021-01-31 by ktbarrett