cocotb/cocotb #526: Assignment with "<=" operator

imphil, 2017-04-24:
The [quickstart tutorial](http://cocotb.readthedocs.io/en/latest/quickstart.html#parallel-and-sequential-execution-of-coroutines) (and other code in the cocotb repository) sometimes uses "signal <= value" to assign values, and other times the regular "signal = value" version. Since I didn't find any documentation (or actually the implementation) on this:

- is there any difference between the two? The usual Verilog blocking/nonblocking assignments cannot be the difference, since the simulator is stopped anyways?
- Is there documentation somewhere, or should it be added?

LinuxFanatic777, 2017-04-24:
I'm using <= and don't know why :) If it works - why bother at all?

AlexanderSpirin, 2017-04-25:
There is no difference between all next assignments, except last one which doesn't really work.
``` python
@cocotb.test()
def parallel_example(dut):
    reset_n = dut.reset
    dut.reset = 1        # OK
    dut.reset <= 1       # OK hdl-like shortcut for the next one 
    dut.reset.value = 1  # OK 
    reset_n <= 1         # OK
    reset_n.value = 1    # OK
    reset_n = 1          # Doesn't work: reference overwrite
```
You can find implementation of all this stuff in handle.py

imphil, 2017-04-25:
Thanks Alexander.

So would it be a good advice to put a simple rule like this into the manual:
"`signal <= new_value`" is a shorthand for `signal.value = new_value`" and adapt the part (http://cocotb.readthedocs.io/en/latest/quickstart.html#assigning-values-to-signals) according to it? (If there's rough agreement on that, I'll go ahead and prepare a PR).

Since the last case you mention (reset_n = 1) makes the use of `signal = new_value" inconsistent depending on the signal being assigned to a variable first, or accessed directly, it confuses people more than it helps. So I'd leave this out of the quickstart guide.

Thanks for the hint to handle.py, the implementation makes this behavior very clear.

luked80, 2017-05-02:
We've tried to stick with `<=` as much as possible for assigning values to simulator signals within cocotb.
For reading back simulator signals, we use `.value` and `int()`, with a slight preference towards `.value`

The ability to pass around signal and module references is powerful, but is one of the gotchas when initially starting to use cocotb, as shown by the example from @AlexanderSpirin above.

juliusbaxter, 2017-05-10:
I'm not sure if this is worth a separate issue (or perhaps has been mentioned before) but I've found that using conditional expressions with the <= assignment can be problematic and strange things occur silently.

For example, as I mentioned in the gitter, something like this will not work:
`bus.sig <= something['sig'] if use_dict else 0`

Basically nothing happens, no assignment occurs leading me to think `__le__` function of the NonHierarchyObject in handles.py never gets called? Worthy of some investigation at some point to help understand this.

However you can work around this by surrounding the right-hand term with parentheses which should cause it to be evaluated first, so something like:
`bus.sig <= (something['sig'] if use_dict else 0`)
... appeared to work fine for me.

closed 2017-05-29 by stuarthodgson

forensicgarlic, 2017-12-27:
i just stumbled across this issue trying to figure out why some code uses <= for signal assignments (like test_mean.py) and some uses = for signal assignments (like test_adder.py). As far as I could find, it wasn't mentioned in the documentation, I had to search the code / issues to figure out they are basically (?) identical.