cocotb/cocotb #527: add PyQt GUI to simulation

LinuxFanatic777, 2017-04-24:


luked80, 2017-05-02:
> When the main loop with 100000 clocks is finished I want to stop clock generation - how to do that?

If you save a reference to your forked coroutine, you can stop it by calling kill. e.g
```
clock_coro = cocotb.fork(clock_100mhz(dut.clock))
........
clock_coro.kill()
```

luked80, 2017-05-02:
> To process Qt events I have to use yield delay(50, units = "us") which is dependent on simulator speed and size of the project - how to make delays in coroutine to be based on real seconds?

I'm not familiar with the Qt event processing loop but what if you just remove the `yield delay` statement in the gui function?

LinuxFanatic777, 2017-05-16:
Here is my example of using PyQt with Cocotb, however this code uses crutch: loopback sockets...

```
import cocotb
from cocotb.triggers import Timer as delay
from cocotb.triggers import RisingEdge as posedge
from cocotb.result import TestFailure as failed

import time, sys
import errno, socket
from multiprocessing import Process
from PyQt4 import QtGui, QtCore, QtNetwork

class gui(QtGui.QDialog):
	def __init__(self, parent = None):
		super(gui, self).__init__(parent)

		layout = QtGui.QVBoxLayout()
		self.setLayout(layout)
		self.label = QtGui.QLabel()
		layout.addWidget(self.label)
		button = QtGui.QPushButton("close")
		button.clicked.connect(self.close)
		layout.addWidget(button)

		self.server = QtNetwork.QTcpServer(self)
		self.server.newConnection.connect(self.new_connection)
		if not self.server.listen(port = 25015):
			self.close()

	def client_read(self):
		text = str(self.client.readAll())
		self.label.setText(text)

	def new_connection(self):
		self.client = self.server.nextPendingConnection()
		self.client.disconnected.connect(self.client.deleteLater)
		self.client.readyRead.connect(self.client_read)

def sim_gui(args):
	app = QtGui.QApplication([])
	_gui = gui()
	_gui.exec_()

@cocotb.test()
def test_gui(dut):
	p_gui = Process(target = sim_gui, args = ("gui",))
	p_gui.start()
	time.sleep(1)
	sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
	sock.connect(("localhost", 25015))

	cocotb.fork(clock_100mhz(dut.clock))
	dut.reset = 1
	for i in range(50):
		yield posedge(dut.clock)
	dut.reset = 0
	for i in range(50000):
		if((i % 10000) == 0):
			try: sock.send(("%s\n" % str(i)).encode("utf-8"))
			except socket.error as serr: pass
		yield posedge(dut.clock)

	try: sock.send(("finished").encode("utf-8"))
	except socket.error as serr: pass
	sock.close()

@cocotb.coroutine
def clock_100mhz(signal):
	period = 10000
	while True:
		signal <= 1
		yield delay(period / 2, units = "ps")
		signal <= 0
		yield delay(period / 2, units = "ps")
```

closed 2017-05-16 by LinuxFanatic777