cocotb/cocotb #528: Is it possible to use Cocotb when RTL compiled by side script without dependencies?

OrionBytes, 2017-04-24:
I want to use Modelsim with Cocotb with customized makefile that will launch Tcl scripts for RTL/Qsys part 
, link Cocotb after. 
Is there any way to do it?
F.e. I know TOPLEVEL module, TOPLEVEL_LANG is verilog and I want to build working solution without
```
ifneq ($(VHDL_SOURCES),)    
	echo "vcom -work $(RTL_LIBRARY) $(VCOM_ARGS) $(VHDL_SOURCES)" >> $@    
endif
ifneq ($(VERILOG_SOURCES),)
	echo "vlog -work $(RTL_LIBRARY) +define+COCOTB_SIM -timescale 1ns/100ps -mfcu +acc=rmb -sv $(VLOG_ARGS) $(VERILOG_SOURCES)" >> $@
endif
```
(from Makefile.questa)

Directly with the
```
echo "vsim $(VSIM_ARGS)" >> $@
echo "source example.tcl" >> $@
echo "run -all" >> $@
```

All my previous attempts were wrong. Is it possible theoretically at all in case of Cocotb?

OrionBytes, 2017-04-24:
I can try the "dirty" way - after all previous steps in Tcl - compile blank RTL file with Cocotb and try to get hierarchy controlled from it. But it's more interesting to explore the way to "connect" Cocotb through the makefiles to the prebuilt design.

ktbarrett, 2021-01-31:
The best way to accomplish this is to copy and modify the cocotb makefiles to your liking. There will be proper documentation depending upon #1718.

ktbarrett, 2024-02-28:
[Covered in docs now.](https://docs.cocotb.org/en/stable/custom_flows.html)

closed 2024-02-28 by ktbarrett