cocotb/cocotb #529: https://gitter.im/cocotb

OrionBytes, 2017-04-25:
I create https://gitter.im/cocotb
to help more people with Cocotb usage and to get help too :)
If @chiggs or anyone will want to create official channel - I will free my.

OrionBytes, 2017-04-25:
@mciepluc @luked80 @leftink and other members without public email on Github - I was unable to invite you.

OrionBytes, 2017-04-25:
@chiggs I add you as administrator if case you prefer to use Gitter too

juliusbaxter, 2017-04-26:
Nice one.

I think it's also worth mentioning that the FOSSi Foundation's LibreCores recently set up a cocotb mailing list: https://lists.librecores.org/listinfo/cocotb

Feel free to sign up and start blasting away there, too.

TC01, 2017-07-24:
It would be nice to add a link to this channel from the README, otherwise people will only stumble across it by looking at the issue tracker.

closed 2018-11-14 by stuarthodgson

eric-wieser, 2019-07-01:
@OrionBytes, can you add @imphil as an administrator too?

PR #674 (2018-11-14): Add link to Gitter channel
Fixes #529.