cocotb/cocotb #530: make[1]: *** [results.xml] Segmentation fault (core dumped) - trying to run example/endian_swapper/tests

Bharath-shashi, 2017-04-30:
Hello,
I'm trying to run the example tests, when I try to make it fails with a segmentation fault. I'm posting the command prompt output. Please take a look. Thank you

user@system:~/Cocotb/examples/endian_swapper/tests$ make
make results.xml
make[1]: Entering directory '/home/user/Cocotb/examples/endian_swapper/tests'
mkdir -p sim_build
cd ../cosim && make
make[2]: Entering directory '/home/user/Cocotb/examples/endian_swapper/cosim'
gcc  -fPIC -pthread -fno-strict-aliasing -O2 -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 \
-fexceptions --param=ssp-buffer-size=4 -D_GNU_SOURCE \
-fwrapv -DNDEBUG -I/usr/include/python2.7 -c io.c -o io_module.o
gcc -pthread -shared  io_module.o -L/usr/lib  -o io_module.so
swig -python -outcurrentdir ../hal/endian_swapper_hal.h
gcc -g -ldl -shared -fPIC -I/home/user/Cocotb/examples/endian_swapper/cosim  -I/usr/include/python2.7 -I../hal ../hal/endian_swapper_hal.c endian_swapper_hal_wrap.c io_module.so -o _hal.so
if [ Linux = Darwin ]; then \
        install_name_tool -change io_module.so /home/user/Cocotb/examples/endian_swapper/cosim/io_module.so _hal.so; \
fi
make[2]: Leaving directory '/home/user/Cocotb/examples/endian_swapper/cosim'
/usr/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1   /home/user/Cocotb/examples/endian_swapper/tests/../hdl/endian_swapper.sv
mkdir -p /home/user/Cocotb/build/obj/x86_64
mkdir -p /home/user/Cocotb/build/libs/x86_64
make -C /home/user/Cocotb/lib/utils SIM=icarus
make[2]: Entering directory '/home/user/Cocotb/lib/utils'
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DICARUS -Wstrict-prototypes -Waggregate-return -DICARUS -c -I/home/user/Cocotb/include -I/usr/include/python2.7  -o /home/user/Cocotb/build/obj/x86_64/cocotb_utils.o cocotb_utils.c
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DICARUS -Wstrict-prototypes -Waggregate-return -DICARUS -shared -Xlinker -L/usr/lib -o /home/user/Cocotb/build/libs/x86_64/libcocotbutils.so /home/user/Cocotb/build/obj/x86_64/cocotb_utils.o
make[2]: Leaving directory '/home/user/Cocotb/lib/utils'
make -C /home/user/Cocotb/lib/gpi_log SIM=icarus
make[2]: Entering directory '/home/user/Cocotb/lib/gpi_log'
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DICARUS -Wstrict-prototypes -Waggregate-return -DFILTER -DICARUS -c -I/home/user/Cocotb/include -I/usr/include/python2.7  -o /home/user/Cocotb/build/obj/x86_64/gpi_logging.o gpi_logging.c
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DICARUS -Wstrict-prototypes -Waggregate-return -DFILTER -DICARUS -shared -Xlinker -L/usr/lib -o /home/user/Cocotb/build/libs/x86_64/libgpilog.so /home/user/Cocotb/build/obj/x86_64/gpi_logging.o   -lpython2.7 -lpthread -ldl  -lutil -lm
make[2]: Leaving directory '/home/user/Cocotb/lib/gpi_log'
make -C /home/user/Cocotb/lib/embed SIM=icarus
make[2]: Entering directory '/home/user/Cocotb/lib/embed'
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DICARUS -Wstrict-prototypes -Waggregate-return -DPYTHON_SO_LIB=libpython2.7.so -DICARUS -c -I/home/user/Cocotb/include -I/usr/include/python2.7  -o /home/user/Cocotb/build/obj/x86_64/gpi_embed.o gpi_embed.c
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DICARUS -Wstrict-prototypes -Waggregate-return -DPYTHON_SO_LIB=libpython2.7.so -DICARUS -shared -Xlinker -L/usr/lib -o /home/user/Cocotb/build/libs/x86_64/libcocotb.so /home/user/Cocotb/build/obj/x86_64/gpi_embed.o   -lpython2.7 -lpthread -ldl  -lutil -lm  -lgpilog -lcocotbutils -L/home/user/Cocotb/build/libs/x86_64
make[2]: Leaving directory '/home/user/Cocotb/lib/embed'
make -C /home/user/Cocotb/lib/gpi EXTRA_LIBS= EXTRA_LIBDIRS= SIM=icarus
make[2]: Entering directory '/home/user/Cocotb/lib/gpi'
g++ -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DVPI_CHECKING -DLIB_EXT=so -DSINGLETON_HANDLES -DICARUS -c -I/home/user/Cocotb/include -I/usr/include/python2.7  -o /home/user/Cocotb/build/obj/x86_64/GpiCbHdl.o GpiCbHdl.cpp
g++ -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DVPI_CHECKING -DLIB_EXT=so -DSINGLETON_HANDLES -DICARUS -c -I/home/user/Cocotb/include -I/usr/include/python2.7  -o /home/user/Cocotb/build/obj/x86_64/GpiCommon.o GpiCommon.cpp
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DVPI_CHECKING -DLIB_EXT=so -DSINGLETON_HANDLES -DICARUS -Wstrict-prototypes -Waggregate-return -DICARUS -shared -Xlinker -L/usr/lib -o /home/user/Cocotb/build/libs/x86_64/libgpi.so  /home/user/Cocotb/build/obj/x86_64/GpiCbHdl.o /home/user/Cocotb/build/obj/x86_64/GpiCommon.o  -lcocotbutils -lgpilog -lcocotb -lstdc++ -L/home/user/Cocotb/build/libs/x86_64
make[2]: Leaving directory '/home/user/Cocotb/lib/gpi'
make -C /home/user/Cocotb/lib/simulator SIM=icarus
make[2]: Entering directory '/home/user/Cocotb/lib/simulator'
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DICARUS -Wstrict-prototypes -Waggregate-return  -DICARUS -c -I/home/user/Cocotb/include -I/usr/include/python2.7  -o /home/user/Cocotb/build/obj/x86_64/simulatormodule.o simulatormodule.c
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DICARUS -Wstrict-prototypes -Waggregate-return  -DICARUS -shared -Xlinker -L/usr/lib -o /home/user/Cocotb/build/libs/x86_64/libsim.so /home/user/Cocotb/build/obj/x86_64/simulatormodule.o   -lgpi -lgpilog -lpython2.7 -lpthread -ldl  -lutil -lm  -L/home/user/Cocotb/build/libs/x86_64
ln -sf libsim.so /home/user/Cocotb/build/libs/x86_64/simulator.so
make[2]: Leaving directory '/home/user/Cocotb/lib/simulator'
make -C /home/user/Cocotb/lib/vpi EXTRA_LIBS= EXTRA_LIBDIRS= SIM=icarus
make[2]: Entering directory '/home/user/Cocotb/lib/vpi'
g++ -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DVPI_CHECKING -DICARUS -c -I/home/user/Cocotb/include -I/usr/include/python2.7  -o /home/user/Cocotb/build/obj/x86_64/VpiImpl.o VpiImpl.cpp
g++ -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DVPI_CHECKING -DICARUS -c -I/home/user/Cocotb/include -I/usr/include/python2.7  -o /home/user/Cocotb/build/obj/x86_64/VpiCbHdl.o VpiCbHdl.cpp
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DVPI_CHECKING -DICARUS -Wstrict-prototypes -Waggregate-return -DICARUS -shared -Xlinker -L/usr/lib -o /home/user/Cocotb/build/libs/x86_64/libvpi.so  /home/user/Cocotb/build/obj/x86_64/VpiImpl.o /home/user/Cocotb/build/obj/x86_64/VpiCbHdl.o   -lgpilog -lgpi -lstdc++  -L/home/user/Cocotb/build/libs/x86_64
ln -sf libvpi.so /home/user/Cocotb/build/libs/x86_64/gpivpi.vpl
ln -sf libvpi.so /home/user/Cocotb/build/libs/x86_64/cocotb.vpi
make[2]: Leaving directory '/home/user/Cocotb/lib/vpi'
PYTHONPATH=/home/user/Cocotb/build/libs/x86_64:/home/user/Cocotb:/home/user/Cocotb/examples/endian_swapper/tests:/home/user/Cocotb/examples/endian_swapper/tests/../cosim: LD_LIBRARY_PATH=/home/user/Cocotb/build/libs/x86_64:/home/user/Cocotb/examples/endian_swapper/tests/../cosim:/home/user/Cocotb/examples/endian_swapper/tests/../cosim:/opt/Xilinx/Vivado/2016.4/lib/lnx64.o:/usr/local/cuda-8.0/lib64:/usr/lib:/usr/lib:/usr/lib:/usr/lib MODULE=test_endian_swapper \
        TESTCASE= TOPLEVEL=endian_swapper_sv TOPLEVEL_LANG=verilog \
        /usr/bin/vvp -M /home/user/Cocotb/build/libs/x86_64 -m gpivpi sim_build/sim.vvp
gpivpi:`/home/user/Cocotb/build/libs/x86_64/gpivpi.vpl' failed to open using dlopen() because:
    /opt/Xilinx/Vivado/2016.4/lib/lnx64.o/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /home/user/Cocotb/build/libs/x86_64/gpivpi.vpl).
/home/user/Cocotb/makefiles/simulators/Makefile.icarus:77: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Segmentation fault (core dumped)
make[1]: Leaving directory '/home/user/Cocotb/examples/endian_swapper/tests'
/home/user/Cocotb/examples/endian_swapper/tests/../../../makefiles/Makefile.sim:71: recipe for target 'sim' failed
make: *** [sim] Error 2

themperek, 2017-04-30:
Remove Xilinx directories from LD_LIBRARY_PATH ? Do not source Xilinx variables?

Bharath-shashi, 2017-04-30:
Thank you,,,, it worked

closed 2017-05-29 by stuarthodgson