cocotb/cocotb #531: changing clock period

kirknodeng, 2017-05-01:
When we generate clock, we use cocotb.coroutine with fork statement. (like below)
Sometimes, I want to change another clock frequency. 

At that time, I think that fork statement should be end and then change frequency...
How can we do that? 

for example : 
def test_change_clock_period(dut) :
    half_period = 1
    cocotb.fork(clock_gen(dut.refclk, half_period))
    # After 20ns change clock
    yield wait_ns(20)
    half_period = 2
    cocotb.fork(clock_gen(dut.refclk, half_period))

@cocotb.coroutine
def clock_gen(signal, period):
    ''' 
    '''
    while True:
        var = 1 
        signal <= 1
        yield wait_ns(period * var)
        signal <= 0
        yield wait_ns(period * var)

luked80, 2017-05-02:
If you save a reference to your forked coroutine, you can kill it when you want to change clock frequency.
e.g
```
fast_clock_coro_ref = cocotb.fork.(clock_gen(dut.refclk,1))
yield Timer(20,units="ns")
fast_clock_coro_ref.kill()
slow_clock_coro_ref = cocotb.fork(clock_gen(dut.refclk,2))
```

stuarthodgson, 2017-05-29:
@kirknodeng please reopen if the response from @luked80 did not point you in the correct direction.

closed 2017-05-29 by stuarthodgson