cocotb/cocotb #534: Ability to disable ansi coloring in simulator transcript

brunopen, 2017-05-05:
Hi,
When running in Questa GUI, if I have a mismatch on the scoreboard, I see a lot of ANSI colouring not properly interpreted in the GUI transcript window. 
It generates lines like : 
# 968.75ns WARNING  cocotb.scoreboard.fs_fpga_top.out         scoreboard.py:137  in compare                         Difference:
#                                                                                                                   [33m0000[39m[49m      28302C20302C2030 2C2030[31m2C[39m[49m[31m20[39m[49m[31m30[39m[49m[31m2C[39m[49m[31m20[39m[49m [35m([39m[49m[35m0[39m[49m[35m,[39m[49m[35m [39m[49m[35m0[39m[49m[35m,[39m[49m[35m [39m[49m[35m0[39m[49m[35m,[39m[49m[35m [39m[49m[35m0[39m[49m[46m[30m,[39m[49m[46m[30m [39m[49m[46m[30m0[39m[49m[46m[30m,[39m[49m[46m[30m [39m[49m

If my understanding is correct, this ANSI content comes from function hexdiffs in utils.py

It would be great if this ANSI would be disabled when COCOTB_ANSI_OUTPUT is set to 0.

Thanks !

stuarthodgson, 2017-05-29:
@brunopen I agree should be easy enough to fix, we are using questa sometimes as well.

cmarqu, 2018-10-25:
This has been fixed by #606, right?

closed 2018-10-26 by stuarthodgson