cocotb/cocotb #536: Memory leak

ajeshp, 2017-05-11:
For simulations with large number of packets streaming in, the memory usage of the simulator seems to continuously grow.
This can be reproduced even in the endian_swapper testbench by increasing the packets to a very large number (300K).
I have a simulation which reads in a pcap which has 5Million messages and icarus seems to consume the whole memory by about 3 million packets.
Any workarounds for this memory leak?

-Ajesh

stuarthodgson, 2017-06-25:
This does seem to be a problem. Just so you know you can run a test with the MEMCHECK=<port> set, then open a browser to localhost:<port>. This uses the dowser and cherrypy modules installable from pip. On the web page it shows the number of objects that are currently in use. If I increase the number of packets to 300000 as you suggest then I see ```__builtin__.dict``` increasing linearly as well as the ```builtin__.generator``` objects as well. I've not looked into if this is the test at fault, the scoreboard or the  coroutine scheduling engine.

ajeshp, 2017-06-26:
Hi Stuart. Thanks for the pointers. I will play around and see if I can find anything. Regards!

AlexanderSpirin, 2017-07-14:
After applying patch #557 memory leak cease in endian_swapper test using Icarus.
But using Modelsim in the same test memory leak unacceptably fast. I found out that one of the problems is VPI handles leak. The root of the promblem is [special define for Modelsim](https://github.com/potentialventures/cocotb/blob/f2ea9cb6a6f365f9b0950ec2aef35293ac806306/lib/vpi/VpiCbHdl.cpp#L107).
@stuarthodgson, can you give some clue why this define is needed for Modelsim. I'am ready to write some code to treat situation accurately. Or mayme it's possible to remove this define? Now I've made so and don't see any errors.
I have a feeling that there are other problems leading to memory leak and it become a big obstacle when you use 32-bit simulator.

ajeshp, 2017-07-14:
I confirm that after applying patch #557 , memory leak issue has been fixed using Icarus.

Thanks @AlexanderSpirin

PR #557 (2017-07-16): Issue #536: Fix memory leak in Scheduler._trigger2coros
After coroutine was unscheduled remove from self._trigger2coros unprimed triggers.
I've checked patch resolves #536
closed 2017-07-16 by stuarthodgson