cocotb/cocotb #537: AXI4Slave Exception :  Write to object m00_axi_WREADY was scheduled during a read-only sync phase.

Martoni, 2017-05-12:
I'm writing a Vivado AXI4 master VHDL component. And to test it I'm using cocotb with amba.py driver to test AXI4.

I had this error when I simulated it : 
```Python
Send raised exception: Write to object m00_axi_WREADY was scheduled during a read-only sync phase.
     File "/opt/cocotb/cocotb/decorators.py", line 115, in send
       return self._coro.send(value)
     File "/opt/cocotb/cocotb/drivers/amba.py", line 248, in _write_data
       self.bus.WREADY <= 1
     File "/opt/cocotb/cocotb/handle.py", line 385, in __le__
       self.value = value
     File "/opt/cocotb/cocotb/handle.py", line 129, in __setattr__
       return object.__setattr__(self, name, value)
     File "/opt/cocotb/cocotb/handle.py", line 401, in <lambda>
       fset=lambda self,v: self._setcachedvalue(v),
     File "/opt/cocotb/cocotb/handle.py", line 585, in _setcachedvalue
       cocotb.scheduler.save_write(self, value)
     File "/opt/cocotb/cocotb/scheduler.py", line 379, in save_write
       raise Exception("Write to object {} was scheduled during a read-only sync phase.".format(handle._name))
```

I manage to correct it with this commit : https://github.com/Martoni/cocotb/commit/dbcb89cd48031c8d1043a7e6aefaeb5ffd493b7f

But not sure that it's the right way to do it.

stuarthodgson, 2017-05-29:
@Martoni You got the error as a user is not allowed to schedule a value chage in ReadOnly(), however this is best way to be certain that the values that are being sampled are stable. What you can do instead is do insert a ```yield NextTimeStep()``` before doing the assignment but after the samling of the condition.

```
         while True:
￼              while True:
￼                  self.bus.WREADY <= 0
￼                  yield ReadOnly()
￼                  if self.bus.AWVALID.value:
+                     yield NextTimeStep()
￼                      self.bus.WREADY <= 1
￼                      break
￼                  yield clock_re
￼  
￼              yield ReadOnly()
```

closed 2017-05-29 by stuarthodgson

Martoni, 2017-05-31:
Hi Stuarthodgson,
I'm using GHDL as simulator, and if I change amba.py as you recommend I got this error : 
https://github.com/tgingold/ghdl/issues/281