cocotb/cocotb #541: Questa/Xilinx Structural Models/Time Resolution

declanca, 2017-05-24:
Folks,

I am having trouble simulating the Xilinx behavioural models for FIFOs. Could be others but only FIFOs right now.

It appears the FLI overrides the Questa Time Resolution to ps.

If I run the cocotb testbench with default time values i.e clock periods in the ps range the Xilinx behavioral models fail to simulate correctly.

If I override the simulator resolution when calling vsim with "-t ns" the simulator time does not advance however the delta count advances. So effectively nothing happens.

If I specify ns clock periods within the testbench the simulations run successfully however they take a lot longer than normal as there are 1000 ps ticks to simulate per ns.

Anyone come across this before?

Thanks!

AlexanderSpirin, 2017-05-25:
> If I run the cocotb testbench with default time values i.e clock periods in the ps range the Xilinx behavioral models fail to simulate correctly.

@declanca, maybe in this case you start to interact with FIFO before 100ns from beginning of simulation and FIFO is effectively in reset by GSR assertion in glbl.v?

> It appears the FLI overrides the Questa Time Resolution to ps.

As I understand simulator choose the minimum time in the whole project and in glbl.v and sim netlists you can find ``` `timescale 1 ps / 1 ps```

declanca, 2017-05-25:
Thanks for the follow up. I am using VHDL therefore had figure it out from that perspective.

After playing with this I figured out that adding a "-t 10ps" to the vsim arguments gets me the performance I need.

And FYI Xilinx behavioural models do not work with 1ps resolution simulations - Mentor is aware of this issue.

stuarthodgson, 2017-05-29:
@declanca this seems not to be a cocotb issue, please reopen if you feel differently.

closed 2017-05-29 by stuarthodgson