cocotb/cocotb #542: hexdiff memory overflow on large transactions

oligu, 2017-06-01:
Around line 250 in utils.py there is a double for-loop traversing two strings  and storing some data in an array:
```python
for j in range(len(y)):
        for i in range(len(x)):
            d[i, j] = ...
```
This is used by the scoreaboard to compare the expected and the received transaction.
In my implementation the length of each of these strings is more than 5000 elements.
This results in an array d[] which is too big to handle and results in a crash of the simulator (ModelSim).

stuarthodgson, 2017-06-25:
This is a really a problem with Modelsim (32 bit by any chance?).

closed 2017-06-25 by stuarthodgson

oligu, 2017-06-27:
ModelSim 32-bit is involved, yes. Have to check if it works with another simulator.