cocotb/cocotb #558: Time in iverilog is measured in seconds

andresdemski, 2017-07-14:
Hello,
I'm starting to use cocotb with iverilog and i found that the arguments of Clock and Timer should be in ps but they are interpreted as seconds.
```
def test (dut):
    cocotb.fork(Clock(dut.clk,100).start())
    yield Timer(100000)
```  
[image]

When i tried to use `units`, it raised the following exception:
[image]

ajeshp, 2017-07-14:
I assume you don't have a `timescale directive declared in dut.
`timescale <reference_time_unit>/<time_resolution>, I usually use `timescale 1ns/1ps

If you don't have it, then adding it should solve your issue.

andresdemski, 2017-07-18:
It works. Thank you

closed 2017-07-18 by andresdemski