cocotb/cocotb #566: Interrupt (control c) causes simulation to freeze rather than exit

TC01, 2017-09-12:
We've been using cocotb with Cadence (ius) for a while now and have found it a really nice way to develop tests!

However we have discovered something that's somewhat frustrating. If I run a simulation and then attempt to interrupt it by pressing control C (that is, by sending a normal Unix interrupt signal to the process), the simulator halts (any output to the terminal stops) but cocotb does not exit; it just sits there for forever.

It's unclear to me why this is the case. My guess is that the simulator dies but some cocotb thread is still running in the background, and so the whole thing refuses to exit.

It would be really nice if instead of hanging, cocotb exited instead of just freezing.

cospan, 2017-09-12:
Have you tried 'Ctrl+\' ?

cospan, 2017-09-12:
Err escape codes!! Ctrl+\\

TC01, 2017-09-13:
> Err escape codes!! Ctrl+\

That does cause the simulation to exit, but I'm not sure what it's doing-- that's not a standard Unix control code, is it? [This page](https://en.wikipedia.org/wiki/C0_and_C1_control_codes) says that ```^\``` is "file separator".

Martoni, 2017-11-03:
I have nearly the opposite problem with icarus. If I press Ctrl+C the simulation pause and I get the icarus console : 
`>>` 
To quit the simulation I have to type «finish» then press enter.

ktbarrett, 2021-01-31:
With a more regular execution flow in cocotb, it is certainly possible to more gracefully end cocotb and signal the simulator to shut down.

cmarqu, 2021-01-31:
`Ctrl-\` will also be in the docs for the 1.5 release (#2343).

ktbarrett, 2025-03-28:
xref #3241 and xref #4585.

ktbarrett, 2025-06-06:
Closed by #4593.

closed 2025-06-06 by ktbarrett