cocotb/cocotb #578: Fix travis builds

stuarthodgson, 2017-12-30:
Travis has now moved on so far that dependencies needed can no longer be found. The travis.yml needs fixing up.

stuarthodgson, 2017-12-30:
This is a touch annoying. Seems that the ppa we were using for obtaining iverilog-daily no longer does a build for even the newest version of ubuntu that is available on travis (14.04) and only goes as far back as 16.04. Short of getting this re-instated (im asking) I'm not currently sure where we go from here.

stuarthodgson, 2017-12-30:
A docker image might a solution here, either a docket image hosted here (but not in this repo) or add a docker file that can be used to build a docker image on the fly.

Another alternative is just to download the iverilog and build from source, but that would chew into the run time we have for tests.

forensicgarlic, 2017-12-31:
This is a battle I've been fighting over the past week. My current choice has been to build iverilog 10.2 from source in travis for a thing I'm working on, and it takes about 6 minutes per build. Building the docker image on the fly is either going to have to start from a base where the iverilog-dailies work, or also build the iverilog source and will take longer than just building iverilog. I haven't investigated hosting a docker image anywhere yet to see if downloading the image is faster than building from source.

imphil, 2018-01-02:
@stuarthodgson For LibreCores CI we've experimented with a couple of "default" Docker images with tools like verilator and iverilog. I'm not completely sure what the current state is, but I think using these images could be a good starting point for the cocotb travis runs. 

@wallento, @oleg-nenashev:
Do we currently have an (continuously updated) Docker image available with icarus verilog and (since this seems to come up repeatedly) GHDL? If not, is this something we could provide and use for LibreCores CI and cocotb testing in travis at the same time? I think this should fit well into the idea of a couple common "base images" to reduce fragmentation.
[If we don't have it right now, I can spend a bit of time on it this week.]

stuarthodgson, 2018-01-03:
@forensicgarlic I think that the iverilog-dailies will work with 16.04 and there seems to be a base docker image available on travis at this version. 

@imphil that sounds like a good path to explore further, assuming that the image does not need to be contained in our repo and can be pulled form anywhere.

Another option is moving the regressions from travis to Libracores CI which I know we have spoken about a bit in the past.

imphil, 2018-01-03:
LibreCores CI is currently rather rough around the edges and for use cases where travis does the job just fine I'd stay with that for now. However, I'll see that we can provide a Docker base image with iverilog and GHDL (and possibly other tools) on DockerHub to be used as base image for the cocotb tests running on travis. This is already planned and partially working, I think only a bit of modification is required to get it ready for cocotb. I'll get back to you on that front in the next couple of days.

imphil, 2018-01-03:
Gave it a try in PR #581, seems to be a reasonable path. @wallento will have a look tomorrow with a bit more input, as he's more familiar with the work done on LibreCores CI. @stuarthodgson it would be nice if you could have a look at the vvp error as noted in the PR.

closed 2018-10-28 by stuarthodgson