cocotb/cocotb #585: Error: undefined reference to `context'

oligu, 2018-01-10:
Since the merge of Pull Request #565 we get following error with ModeSim 32-bit on Windows:

```Console
make -C /e/projects/EDA/Python/cocotb/lib/utils SIM=modelsim
make[2]: Entering directory `/e/projects/EDA/Python/cocotb/lib/utils'
gcc -m32 -g -DDEBUG -shared -DMODELSIM -c -I/e/projects/EDA/Python/cocotb/include -Ic:\Python27\include  -o /e/projects/EDA/Python/cocotb/build/obj/i686/cocotb_utils.o cocotb_utils.cgcc -m32 -g -DDEBUG -shared -DMODELSIM -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Python27/libs -o /e/projects/EDA/Python/cocotb/build/libs/i686/libcocotbutils.dll /e/projects/
EDA/Python/cocotb/build/obj/i686/cocotb_utils.oe:/projects/EDA/Python/cocotb/build/obj/i686/cocotb_utils.o: In function `to_python':
e:/projects/EDA/Python/cocotb/include/cocotb_utils.h:49: undefined reference to `context'
e:/projects/EDA/Python/cocotb/include/cocotb_utils.h:53: undefined reference to `context'
e:/projects/EDA/Python/cocotb/include/cocotb_utils.h:53: undefined reference to `context'
e:/projects/EDA/Python/cocotb/build/obj/i686/cocotb_utils.o: In function `to_simulator':
e:/projects/EDA/Python/cocotb/include/cocotb_utils.h:58: undefined reference to `context'
e:/projects/EDA/Python/cocotb/include/cocotb_utils.h:63: undefined reference to `context'
e:/projects/EDA/Python/cocotb/build/obj/i686/cocotb_utils.o:e:/projects/EDA/Python/cocotb/include/cocotb_utils.h:63: more undefined references to `context' follow
collect2.exe: error: ld returned 1 exit status
make[2]: *** [/e/projects/EDA/Python/cocotb/build/libs/i686/libcocotbutils.dll] Error 1
make[2]: Leaving directory `/e/projects/EDA/Python/cocotb/lib/utils'
make[1]: *** [/e/projects/EDA/Python/cocotb/build/libs/i686/libcocotbutils.dll] Error 2
make[1]: Leaving directory `/e/projects/CubeSatCOM/dev/verificaton/cocotb/HEI/CubeSatCOM/Channel/ReedSolomonEncoder'
make: *** [sim] Error 2
```

imphil, 2018-01-12:
The `context` variable should be set in lib/embed/gpi_embed.c, i.e. the part of cocotb loaded into the simulator. Could it be that this library wasn't rebuilt when you updated cocotb? Can you try deleting all build outputs and re-try with a clean environment?

oligu, 2018-01-12:
Thanks for the reply. I did a ```make clean``` without any success. Do I need to do something more to delete all building outputs?

Looking at the output of the last compilation that runs without that error, it looks like ```cocotb_utils``` gets compiled before ```gpi_embed```. So the compiler/linker might not know yet that variable.
```Console
$ make clean; make SIM=modelsim TOPLEVEL_LANG=vhdl COCOTB_ANSI_OUTPUT=1 COCOTB_REDUCED_LOG_FMT=1
make results.xml
make[1]: Entering directory `/e/projects/CubeSatCOM/dev/verificaton/cocotb/HEI/CubeSatCOM/Channel/CommunicationsLinkTransmissionUnit'
echo "if [file exists work] {vdel -lib work -all}" > sim_build/runsim.do
echo "vlib work" >> sim_build/runsim.do
echo "vmap -c" >> sim_build/runsim.do
echo "vmap work work" >> sim_build/runsim.do
echo "vcom -work work  e:/projects/CubeSatCOM/dev/verificaton/cocotb/HEI/CubeSatCOM/Channel/CommunicationsLinkTransmissionUnit/../../../../../../implementation/spinal/SpinalBaseProject/CommunicationsLinkTransmissionUnit.vhd"
>> sim_build/runsim.do
echo "vsim -onfinish exit -foreign \"cocotb_init libfli.dll\" work.communicationslinktransmissionunit " >> sim_build/runsim.do
echo "onbreak resume" >> sim_build/runsim.do
echo "run -all" >> sim_build/runsim.do
echo "quit" >> sim_build/runsim.do
mkdir -p /e/projects/EDA/Python/cocotb/build/obj/i686
mkdir -p /e/projects/EDA/Python/cocotb/build/libs/i686
make -C /e/projects/EDA/Python/cocotb/lib/utils SIM=modelsim
make[2]: Entering directory `/e/projects/EDA/Python/cocotb/lib/utils'
gcc -m32 -g -DDEBUG -shared -DMODELSIM -c -I/e/projects/EDA/Python/cocotb/include -Ic:\Python27\include  -o /e/projects/EDA/Python/cocotb/build/obj/i686/cocotb_utils.o cocotb_utils.c
gcc -m32 -g -DDEBUG -shared -DMODELSIM -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Python27/libs -o /e/projects/EDA/Python/cocotb/build/libs/i686/libcocotbutils.dll /e/projects/
EDA/Python/cocotb/build/obj/i686/cocotb_utils.o
make[2]: Leaving directory `/e/projects/EDA/Python/cocotb/lib/utils'
make -C /e/projects/EDA/Python/cocotb/lib/gpi_log SIM=modelsim
make[2]: Entering directory `/e/projects/EDA/Python/cocotb/lib/gpi_log'
gcc -m32 -g -DDEBUG -shared -DFILTER -DMODELSIM -c -I/e/projects/EDA/Python/cocotb/include -Ic:\Python27\include  -o /e/projects/EDA/Python/cocotb/build/obj/i686/gpi_logging.o gpi_logging.c
gcc -m32 -g -DDEBUG -shared -DFILTER -DMODELSIM -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Python27/libs -o /e/projects/EDA/Python/cocotb/build/libs/i686/libgpilog.dll /e/proje
cts/EDA/Python/cocotb/build/obj/i686/gpi_logging.o   -lPython27
make[2]: Leaving directory `/e/projects/EDA/Python/cocotb/lib/gpi_log'
make -C /e/projects/EDA/Python/cocotb/lib/embed SIM=modelsim
make[2]: Entering directory `/e/projects/EDA/Python/cocotb/lib/embed'
gcc -m32 -g -DDEBUG -shared -DPYTHON_SO_LIB=Python27.dll -DMODELSIM -c -I/e/projects/EDA/Python/cocotb/include -Ic:\Python27\include  -o /e/projects/EDA/Python/cocotb/build/obj/i686/gpi_embed.o gpi_embed.c
gpi_embed.c: In function 'embed_init_python':
gpi_embed.c:108:81: warning: implicit declaration of function 'getpid' [-Wimplicit-function-declaration]
         fprintf(stderr, "Waiting for %lu seconds - Attach to %d\n", sleep_time, getpid());
                                                                                 ^
gcc -m32 -g -DDEBUG -shared -DPYTHON_SO_LIB=Python27.dll -DMODELSIM -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Python27/libs -o /e/projects/EDA/Python/cocotb/build/libs/i686/li
bcocotb.dll /e/projects/EDA/Python/cocotb/build/obj/i686/gpi_embed.o   -lPython27 -lgpilog -lcocotbutils -L/e/projects/EDA/Python/cocotb/build/libs/i686
make[2]: Leaving directory `/e/projects/EDA/Python/cocotb/lib/embed'
```

flekilloloko, 2018-01-19:
I have the same 'context' error msg. Can't find a workaround yet.

pichi-loco, 2018-03-07:
A roll back to the 6c432ee73 commit solved the issue
here is the link to the repo at that time
https://github.com/potentialventures/cocotb/tree/6c4320ee732b2af11faee17de57b9f40a4a5662c

imphil, 2018-03-07:
So this happened most likely in this merge: https://github.com/potentialventures/cocotb/commit/f76e4fc43cff071d84a86b9c35529577df57473a If you find time, could you bisect the commits in this merge to see which commit is actually at fault? (And first verify that it works before this merge, and not after.)

elgorwi, 2018-06-02:
@oligu I've  issued #620. This should get around the compilation error.

oligu, 2018-06-04:
@elgorwi thanks a lot. indeed, this solves the error.

closed 2018-10-28 by stuarthodgson