cocotb/cocotb #587: A way to call non-standard system tasks?

gradrat, 2018-01-12:
I/We are just starting to give cocotb a try and like what we see so far.  

One hurdle, however, is that we'd like to be able to call and capture the output of non-standard system tasks that might be available/registered for a specific simulator or environment. 

For example, we often run simulations in a mixed-signal environment where half the design may be RTL and the other half is running some SPICE-like simulator. In these cases, we've been using the simulator's custom system tasks to capture the value of certain analog signals that we are interested in.

I'm not too familiar with Verilog PLI/VPI (so forgive me for any misconceptions here), but it seems as if interfaces to these custom system tasks could be added to the supporting C & Python libraries in cocotb. However, what would be more ideal and universally usable is if there were a general way to call an arbitrarily named system task in Verilog through some cocotb method. Is this possible?

cmarqu, 2018-10-26:
At least for Cadence simulators, you can use a "PLI/VPI Map File" which allows you to map system functions to VPI functions. I have no experience with that though.

cmarqu, 2020-08-20:
Related: #1051

ktbarrett, 2020-12-18:
The VPI does not support the ability to call tasks or functions defined in Verilog from C. The DPI can do this. See #1989.

closed 2020-12-18 by ktbarrett