cocotb/cocotb #595: Wrong include dir syntax for VCS/Questa/IUS in axi_lite_slave example

pulzar, 2018-02-13:
The Makefile in axi_lite_slave example sets the include dir for HDL code using -I<path> syntax used by gcc, but VCS/Questa/IUS (and likely other simulators that I can't try) require +incdir+<path> syntax instead.

The following change is needed to make this example work:

```
diff --git a/examples/axi_lite_slave/tests/Makefile b/examples/axi_lite_slave/tests/Makefile
index 4d2e8f4..e8bded7 100644
--- a/examples/axi_lite_slave/tests/Makefile
+++ b/examples/axi_lite_slave/tests/Makefile
@@ -8,7 +8,7 @@ PYTHONPATH := ./model:$(PYTHONPATH)
 export PYTHONPATH
 export PYTHONHOME=$(shell python -c "from distutils.sysconfig import get_config_var; print(get_config_var('prefix'))")

-EXTRA_ARGS+=-I$(TOPDIR)/hdl/
+EXTRA_ARGS+=+incdir+$(TOPDIR)/hdl/

```

cmarqu, 2018-10-24:
I guess those test specific Makefiles should not actually fill in ``EXTRA_ARGS`` but rather set some list variable like ``INCDIRS`` what can then be prefixed with the right option in each of the simulator-specific Makefiles.

ktbarrett, 2021-01-31:
Fixed in https://github.com/cocotb/cocotb/commit/11bd4391a195acb0088a1e9bfcb01d9359bae34b.

closed 2021-01-31 by ktbarrett