cocotb/cocotb #602: Questa/Modelsim: don't force compilation unit settings

tovine, 2018-03-06:
I discovered an issue today while trying to simulate a medium-sized project (several files) which uses SystemVerilog packages for constant and property definitions; it complained about multiple declarations of the same properties through wildcard imports.

```
# ** Error: /path/to/file.sv(line): Identifier 'C_ABCDE' is not directly visible.
#    Found multiple Declaration of 'C_ABCDE' through wildcard imports from these packages : component_defs, component_defs, component_defs, component_defs, component_defs, component_defs, component_defs, component_defs, component_defs, component_defs, component_defs, component_defs
```

I fixed this by modifying the `cocotb/makefiles/simulators/Makefile.questa` file to remove the `-mfcu` option at line 98, allowing QuestaSim to treat each file as a separate compile unit (using the `-sfcu` option instead). If you want I could submit a pull request with this change.

forensicgarlic, 2018-03-06:
This can pretty easily cause issues in projects where they assume -mfcu is used, and don't import a package in a file where it's used. I think this is not great design, but I also know my last workplace made this exact assumption in their projects requiring -mfcu. 

I don't know if the same can be said for wildcard imports and needing -sfcu. 

Perhaps instead the project makefile should set 
`
FCU=-sfcu`

and the simulator makefile can set 
`
FCU?=-mfcu`

and line 98 can be 
`
	@echo "vlog -work $(RTL_LIBRARY) +define+COCOTB_SIM -timescale 1ns/100ps $(FCU) +acc=rmb -sv $(VLOG_ARGS) $(VERILOG_SOURCES)" >> $@`

to preserve legacy functionality?

tovine, 2018-03-07:
That is possible, but I feel filthy just thinking about it - assuming a
default that is even opposite of the simulator default (at least for
Questa) is in my opinion a ticking bomb.

I absolutely get your point (I have also struggled with workarounds due to
legacy software decisions), but would also argue that it won't
*immediately* cause problems to existing users as they would first have to
pull the newer version from git (and as with any software update there may
be regressions which they should be aware of). If it is explicitly stated
in release notes (or at least commit messages) then it should be fairly
quick to figure out what went wrong and add the flag when you get the error
message, no?

Your proposal to use the FCU variable with default to `-mfcu` seems like a
reasonable compromise if one *must* ensure legacy compatibility with risky
assumptions, but personally I would just remove the flag from the line
altogether and have it passed in VLOG_ARGS if needed.

PS: technically timescale could also be an option, but that's a bit
off-topic here

>

mciepluc, 2018-03-07:
1. I think eventually VLOG_ARGS/VCOM_ARGS must be replaced with COMPILE_ARGS and VSIM_ARGS with SIM_ARGS, to ensure consistency between Make scripts.
2. Introducing additional FCU will be a bit messy in my opinion, as this option will be useful only for Questa.
3. Maybe some solution is the following: add variable containing default args, such as DEFAULT_COMPILE_ARGS:
```DEFAULT_COMPILE_ARGS= -timescale 1ns/100ps -mfcu +acc=rmb -sv```
and then compilation call will be:
```@echo "vlog -work $(RTL_LIBRARY) +define+COCOTB_SIM $(DEFAULT_COMPILE_ARGS) $(COMPILE_ARGS) $(VERILOG_SOURCES)" >> $@```
So basically, if you need to replace default settings you should override the DEFAULT_COMPILE_ARGS. If you need to pass additional settings, you just define COMPILE_ARGS.
The question is whether there are different compilation args needed for VHDL (vcom) compilation or the same variables (DEFAULT_COMPILE_ARGS, COMPILE_ARGS) could be reused.

Please also have a look at the #291

forensicgarlic, 2018-03-07:
you're right @mciepluc, that's a better, more global sort of fix. 

and @tovine, I do think it's fine to ask "How come the defaults are set to this" as well as asking for an easier way for a project to override defaults. I'm just not personally sure of the answer to the first, which is why I went for an answer to the second.

tovine, 2018-03-07:
@mciepluc Just to clarify: I did not set `VSIM_ARGS`, it was correctly applied from the `EXTRA_ARGS` variable in the makefile so that problem has already been solved.

Using overridable `DEFAULT_COMPILE_ARGS` could be an idea.

@forensicgarlic I agree, it's a mystery to me why it was deemed necessary to set the defaults like this - possibly to ensure consistent behavior across different compiler/simulator backends?

However it is a bit problematic as long as the compiler freaks out if both opposing arguments are applied at the same time, failing to execute instead of allowing the user to override the (perhaps wrongfully assumed) cocotb defaults.

cmarqu, 2018-11-06:
>    So basically, if you need to replace default settings you should override the DEFAULT_COMPILE_ARGS. If you need to pass additional settings, you just define COMPILE_ARGS.

One small UX comment here - intuitively, I would have assumed that just by setting ``COMPILE_ARGS``, I would completely override ``DEFAULT_COMPILE_ARGS``, while your code example does not do this.
Of course, one can get used to it, but this may be a small trap for users.