cocotb/cocotb #603: disable ANSI output in GUIs

s3bs, 2018-03-09:
Colored output in the console is an awesome feature but in the riviera GUI not so much, because it prints the format string instead. 

```
Difference:

# COUT:                                                                                                                    [36m0000[39m[49m [36m0000[39m[49m 466972656420626C 6164653D31206261 [35mF[39m[49m[35mi[39m[49m[35mr[39m[49m[35me[39m[49m[35md[39m[49m[35m [39m[49m[35mb[39m[49m[35ml[39m[49m[35ma[39m[49m[35md[39m[49m[35me[39m[49m[35m=[39m[49m[35m1[39m[49m[35m [39m[49m[35mb[39m[49m[35ma[39m[49m

# COUT:                                                                                                                     [36m0010[39m[49m [36m0010[39m[49m 7463683D31206174 207061796C6F6164 [35mt[39m[49m[35mc[39m[49m[35mh[39m[49m[35m=[39m[49m[35m1[39m[49m[35m [39m[49m[35ma[39m[49m[35mt[39m[49m[35m [39m[49m[35mp[39m[49m[35ma[39m[49m[35my[39m[49m[35ml[39m[49m[35mo[39m[49m[35ma[39m[49m[35md[39m[49m

# COUT:                                                                                                                     [33m0020[39m[49m      5F64657074683D34 [31m30[39m[49m                      [35m_[39m[49m[35md[39m[49m[35me[39m[49m[35mp[39m[49m[35mt[39m[49m[35mh[39m[49m[35m=[39m[49m[35m4[39m[49m[46m[30m0[39m[49m

# COUT:                                                                                                                          [33m0020[39m[49m 5F64657074683D34 [31m34[39m[49m                      [35m_[39m[49m[35md[39m[49m[35me[39m[49m[35mp[39m[49m[35mt[39m[49m[35mh[39m[49m[35m=[39m[49m[35m4[39m[49m[46m[30m4[39m[49m

# COUT:                                                                                                                     
```

both the logger and the hexdiffs are affected by this.

forensicgarlic, 2018-03-09:
Have you already tried adding this to your makefile? 
`
COCOTB_ANSI_OUTPUT=0`

luked80, 2018-03-13:
You can just add COCOTB_ANSI_OUTPUT=0 to your command line e.g
`make COCOTB_ANSI_OUTPUT=0 run_riveria`

s3bs, 2018-03-14:
@luked80 

> More importantly and that's what my example above shows is, that the output produced by the scoreboard hexdiff doesn't even respect COCOTB_ANSI_OUTPUT.

luked80, 2018-03-14:
@s3bs ah ok, I see the problem.
Just pushed a fix and opened a pull request.

s3bs, 2018-03-15:
@luked80 thanks.

closed 2018-10-28 by stuarthodgson