cocotb/cocotb #607: debug Modelsim under Windows 7 64bits

svancau, 2018-03-20:
Hello all,
I have issues with simulation with cocotb and Modelsim under Windows 7 64 bits. In gui mode, Modelsim fails big time doing a segfault, whereas in console mode there is no problem.

```
/c/questasim64_10.5c/win64/vsim -gui -64 -do runsim.do 2>&1 | tee sim.log
# Attempting stack trace sig 11
# Signal caught: signo [11]
# vsim_stacktrace.vstf written
# Current time Tue Mar 20 09:06:21 2018
# Program = vsim
# Id = "10.5c"
# Version = "2016.07"
# Date = "Jul 21 2016"
# Platform = win64
# Signature = 2f6a33b7a487e2d6759e4152b9ebdbb7
# End of Stack Trace

** Fatal: (SIGSEGV) Bad pointer access. Closing vsimk.
** Fatal: vsimk is exiting with code 211.
Exit codes are defined in the "Error and Warning Messages"
appendix of the QuestaSim User's Manual.
make[1] : on quitte le répertoire « /c/Projects/DHR/A000/SupportModules/hwprot_tb/cocotb »
```

How would you debug that issue ? any ideas, ? I don't know where to begin.
Thanks

luked80, 2018-03-21:
What does vsim_stacktrace.vstf show?

Can you run under gdb or valgrind?

Is it possible for someone to recreate this?

oligu, 2018-03-21:
Hi Sebastien
Do you have the same issue also with the 32-bit version?

closed 2018-07-29 by stuarthodgson