cocotb/cocotb #608: wrong BinaryValue assign/initialization with 3 bits

Fatsie, 2018-04-02:
Following code:
```#!python
    v = BinaryValue(3,3)
    print(v.integer)
```
gives 6 as output
This is under python 3.6

wstucker, 2018-04-05:
@Fatsie  Your example used the default kwarg of bigEndian=True. I think that setting bigEndian=False would probably give you the expected behavior.

```
be = BinaryValue(3,3)
print("Big Endian", be.integer)
le = BinaryValue(3,3, bigEndian=False)
print("Little Endian", le.integer)
```

Yields:

```
('Big Endian', 6)
('Little Endian', 3)
```

Fatsie, 2018-04-08:
> 5493e4bb","name":"GitHub"},"entity":{"external_key":"github/potential
> ventures/cocotb","title":"potentialventures/cocotb","subtitle":"GitHu
> b repository","main_image_url":"https://cloud.githubusercontent.com/a
> ssets/143418/17495839/a5054eac-5d88-11e6-95fc-
> 7290892c7bb5.png","avatar_image_url":"https://cloud.githubusercontent
> .com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-
> b52498112777.png","action":{"name":"Open in
> GitHub","url":"https://github.com/potentialventures/cocotb"}},"update
> s":{"snippets":[{"icon":"PERSON","message":"@wstucker in #608:
> expected behavior.\r\n\r\n```\r\nbe = BinaryValue(3,3)\r\nprint(\"Big
> Endian\", be.integer)\r\nle = BinaryValue(3,3,
> bigEndian=False)\r\nprint(\"Little Endian\",
> le.integer)\r\n```\r\n\r\nYields:\r\n\r\n```\r\n('Big Endian',
> 6)\r\n('Little Endian', 3)\r\n```"}],"action":{"name":"View
> Issue","url":"https://github.com/potentialventures/cocotb/issues/608#
> issuecomment-379104050"}}}

I can't see this reply online so I reply by email.
OK, but I think this still indicates inconsistency with how bigEndian
value is treated during initialization and conversion back to integer.
greets,Staf.

Fatsie, 2018-04-08:
Did some more testing with results given below.

init int | init length | init bigEndian | .binstr | .integer
-------- | ----------- | ------------------ | -------- | ---------
3 | 3 | True | 110 | 6
4 | 3 | True | 100 | 4
3 | 3 | False | 011 | 3
4 | 3 | False | 100 | 4
5 | 5 | True | 10100 | 20
2 | 5 | True | 10000 | 16
5 | 5 | False | 00101 | 5
2 | 5 | False | 00010 | 2

So to me the bug seems to be that conversion from and to integer is wrong for bigEndian=True

eric-wieser, 2018-07-31:
Seems I re-reported this as #636

hashhsah, 2019-02-12:
I hit this bug, too. Looking forward to its fix being merged.

In addition, I wonder `if big_endian` could be a class variable so I can set endianess globally in a project. This way, I would still have the option of setting the opposite endianess to a few `BinaryValue` object.

juliusbaxter, 2019-02-13:
I'd second exposing the BinaryValue default endianess as a global switch which can be set.

I'm not sure how Pythonic that is, or whether this would result in compatibility issues, for example if one module sets it one way, and the next sets it back, would that break functionality in the former?

aubindetrez, 2022-07-13:
I started working on a fix for this binary.BinaryValue issue regarding BigEndian.

Each BinaryRepresentation needs to be tested/fixed:
- [x] UNSIGNED
- [ ] SIGNED_MAGNITUDE
- [ ] TWOS_COMPLEMENT

With more testing we may find other issue with that file...
I am thinking about openning a PR when every BinaryRepresentations are tested.

imphil, 2022-07-13:
@aubindetrez I know @ktbarrett has been thinking about BinaryValue a lot, but I think nobody ever found the time to push it through completely. Would you be interested in spending a bit time on that? It's not a 1 day-project, but could be quite rewarding I think. Let me know!

aubindetrez, 2022-07-14:
@imphil Yes I can work on it.

If BinaryValue has to stay around for compatibility reason I think it is worth fixing what it really wrong/broken.
Meanwhile maybe we can flag it as 'deprecated' and propose a replacement? (like the mentioned `cocotb.types.LogicArray`)

ktbarrett, 2022-07-14:
1.7 will be released soon. Then I think we should start breaking things wherever we see fit.

ktbarrett, 2024-05-18:
BinaryValue is dead.

closed 2024-05-18 by ktbarrett