cocotb/cocotb #610: Suggestion: Switch to `async` based coroutines instead of yield-based

eric-wieser, 2018-04-24:
Python 3 has native language support for coroutines, it would be a shame not to use it

nevsan, 2018-04-24:
What's the thought on Python compatibility going forward? A lot of things could probably be cleaned up by dropping Python 2.7 support completely.

stuarthodgson, 2018-05-29:
Python 2.7 wont be going away just yet, there are too many users, us for one. This could be the big factor in moving as we can remove a lot of code from the repo but other parts will need factoring again as well so I think its a good chunk of work and will mean we no longer support 2.7 as I dont fancy maintaining two separate schedulers.

eric-wieser, 2018-07-28:
Turns out you can get `await` syntax with something as simple as

```
Trigger.__await__ = lambda self: (yield self)
RunningCoroutine.__await__ = lambda self: (yield self)
```

eric-wieser, 2018-07-29:
Implementing the above works around #632 by letting python handle it natively

stuarthodgson, 2018-07-29:
The part that really needs figuring out is allowing existing test benches that use yield to work seemlessly with Async/await internally. Without this then the migration forwards is too painful and so the existing mechanism won't just be removed for the sake of moving to python 3.

eric-wieser, 2018-07-29:
I'm not sure what you mean by using async / await internally. In the example above, I demonstrate that new-style async functions can both call and be called by old-style cocotb.coroutines

stuarthodgson, 2018-07-29:
I mean that any implementation of this will need to allow existing code in test benches to work with no modifications. What is your plan for making the existing test cases work with the await internal scheduler. What are your goals with the work it's not clear to me how far you plan to go.

eric-wieser, 2018-07-29:
So to be clear, there are two separate requests in this PR:
1. Add support for using the `async` / `await` syntax (#633), which:
  * enables `async for`, `async with`
  * Lets the language handle nested calls to coroutines:
    * Fixes #632 
    * Gives better stack traces
2. Leverage parts of `asyncio` to build the internal scheduler
  * Either requires dropping python <3.5 (obviously not viable yet), or adding a dependency on `trollius`, which is a backport of `asyncio`

As a user, I only care about 1 - but I think 2 could reduce maintenance work down the line

cmarqu, 2018-11-25:
Any thoughts on ``trio``, even though it's not a standard library?
(Since it claims to "solve the Ctrl-C problem", maybe it also helps with #566.)

imphil, 2019-06-27:
@eric-wieser all actionable items from this issue are resolved in current master, aren't they? (The asyncio part isn't covered, but that's not really in the title of the issue either.)

eric-wieser, 2019-07-14:
@imphil: I think while point 2 above is still actionable, it doesn't really fit under the title of this issue - so fine to close

closed 2019-07-14 by eric-wieser