cocotb/cocotb #612: How to run with IES 15.20

awakeford, 2018-05-14:
Hi, I am trying to run with IES 15.20 but am not able to make due to an ELF header error.

I am using python 3.5.2 on CentOS. Our system has multiple python versions installed and we specify the version using "setenv PKGVERSION_python 3.5.2", I am not sure if you are familiar with that setup but I am not the SU here so I don't have all the info on how the python version is installed.

This is the makefile pythong debug output:

```
Debug from Makefile.pylib
	Using PYTHON_VERSION=3.5
	Using PYTHON_LIBDIR=/usr/opt/pmc/python/python-3.5.2.x86_64/lib
	Using PYTHON_DYN_LIB=libpython3.5m.a
	Using PYTHON_DYNLIBDIR=/usr/opt/pmc/python/python-3.5.2.x86_64/lib/python3.5/lib-dynload
	Using PYTHON_INCLUDEDIR=/usr/opt/pmc/python/python-3.5.2.x86_64/include/python3.5m
file /usr/opt/pmc/python/python-3.5.2.x86_64/lib/libpython3.5m.a
/usr/opt/pmc/python/python-3.5.2.x86_64/lib/libpython3.5m.a: current ar archive
file -L /usr/opt/pmc/python/python-3.5.2.x86_64/lib/libpython3.5m.a
/usr/opt/pmc/python/python-3.5.2.x86_64/lib/libpython3.5m.a: current ar archive
```

When I try and make I get the following:

```
Failed to find python lib
Unable to open lib libpython3.5m.a (/usr/opt/pmc/python/python-3.5.2.x86_64/lib/libpython3.5m.a: invalid ELF header)
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
ncsim> source /tools/cds/ies/ies15.20.024.lnx86/tools/inca/files/presrc.tcl
NCM: Mon May 14 10:31:24 PDT 2018 1v1526319084sjc2f10ac959p29188 Host=sjc2virdt200 JobID=n/a (cad_dd_00989)
Simulation SnapShot is worklib.endian_swapper_sv:sv

ncsim> source /tools/cds/ies/ies15.20.024.lnx86/tools/inca/files/ncsimrc
ncsim> run
Traceback (most recent call last):
  File "/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/cocotb/__init__.py", line 35, in <module>
    import logging
  File "/usr/lib64/python2.6/logging/__init__.py", line 34, in <module>
    import sys, os, types, time, string, cStringIO, traceback
ImportError: /usr/lib64/python2.6/lib-dynload/timemodule.so: undefined symbol: PyExc_ValueError
```

I unarchived one of the object files from libpython3.5m.a, and checked that it was indeed 64bit. 

```
ar -x /usr/opt/pmc/python/python-3.5.2.x86_64/lib/libpython3.5m.a bytesio.o
file bytesio.o
bytesio.o: ELF 64-bit LSB relocatable, x86-64, version 1 (SYSV), not stripped
```

Doing some tests, I also confirmed that I am using 64bit IUS with 64bit python.

`> /tools/cds/ies/ies15.20.024.lnx86/tools/inca/bin/64bit/irun: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 2.6.9, stripped`

```
python -c 'import ctypes;print (ctypes.sizeof(ctypes.c_voidp))'
8
```
i.e. 8bytes = 64bit

Any ideas? 
Thanks a lot.

luked80, 2018-05-15:
I have run cocotb with python 3.5 on CentOS 5 and 6 with 64-bit IES 15.
Hard to debug much further without being able to recreate your failing example.
Could you provide a testcase or instructions on how to recreate?

luked80, 2018-05-15:
Seems like you are using irun.
An obvious one but are you invoking with "-64bit"?

awakeford, 2018-05-15:
Thanks for responding Luked80,

I noticed that the irun call was "irun -64" and changed this to "irun -64bit" in Makefile.ius instead but this did not change anything. 

I am actually trying to run the endian_swapper test right off a cocotb fresh install.

I am checking out from Github, 
cd into cocotb/examples/endian_swapper/tests
make SIM=ius

The irun command line is:

`/tools/cds/ies/ies15.20.024.lnx86/tools/bin/64bit/irun -64bit -nclibdirpath sim_build -plinowarn -sv -v93 -loadvpi /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/build/libs/x86_64/libvpi +access+rwc /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/examples/endian_swapper/tests/../hdl/endian_swapper.sv `

Here is my full log:
-------------------------------------------------------------------------------------------------------
```
/usr/bin/make results.xml
make[1]: Entering directory `/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/examples/endian_swapper/tests'
mkdir -p sim_build
cd ../cosim && make
make[2]: Entering directory `/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/examples/endian_swapper/cosim'
gcc  -fPIC -pthread -fno-strict-aliasing -O2 -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 \
	-fexceptions --param=ssp-buffer-size=4 -D_GNU_SOURCE \
	-fwrapv -DNDEBUG -I/usr/opt/pmc/python/python-3.5.2.x86_64/include/python3.5m -c io.c -o io_module.o
io.c: In function 'initio_module':
io.c:120: warning: implicit declaration of function 'Py_InitModule'
io.c:120: warning: assignment makes pointer from integer without a cast
io.c:125: warning: control reaches end of non-void function
gcc -pthread -shared  io_module.o -L/usr/opt/pmc/python/python-3.5.2.x86_64/lib  -o io_module.so
swig -python -outcurrentdir ../hal/endian_swapper_hal.h
gcc -g -ldl -shared -fPIC -I/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/examples/endian_swapper/cosim  -I/usr/opt/pmc/python/python-3.5.2.x86_64/include/python3.5m -I../hal ../hal/endian_swapper_hal.c endian_swapper_hal_wrap.c io_module.so -o _hal.so
endian_swapper_hal_wrap.c: In function 'SWIG_Python_GetModule':
endian_swapper_hal_wrap.c:2428: warning: assignment makes pointer from integer without a cast
endian_swapper_hal_wrap.c: In function 'SWIG_Python_SetModule':
endian_swapper_hal_wrap.c:2497: warning: initialization makes pointer from integer without a cast
endian_swapper_hal_wrap.c: In function 'SWIG_Python_TypeQuery':
endian_swapper_hal_wrap.c:2520: warning: cast to pointer from integer of different size
endian_swapper_hal_wrap.c:2525: warning: assignment makes pointer from integer without a cast
if [ Linux = Darwin ]; then \
		install_name_tool -change io_module.so /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/examples/endian_swapper/cosim/io_module.so _hal.so; \
	fi
make[2]: Leaving directory `/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/examples/endian_swapper/cosim'
mkdir -p /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/build/obj/x86_64
mkdir -p /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/build/libs/x86_64
make -C /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/lib/utils SIM=ius
make[2]: Entering directory `/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/lib/utils'
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DIUS -Wstrict-prototypes -Waggregate-return -DIUS -c -I/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/include -I/usr/opt/pmc/python/python-3.5.2.x86_64/include/python3.5m  -o /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/build/obj/x86_64/cocotb_utils.o cocotb_utils.c
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DIUS -Wstrict-prototypes -Waggregate-return -DIUS -shared -Xlinker -L/usr/opt/pmc/python/python-3.5.2.x86_64/lib -o /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/build/libs/x86_64/libcocotbutils.so /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/build/obj/x86_64/cocotb_utils.o    
make[2]: Leaving directory `/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/lib/utils'
make -C /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/lib/gpi_log SIM=ius
make[2]: Entering directory `/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/lib/gpi_log'
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DIUS -Wstrict-prototypes -Waggregate-return -DFILTER -DIUS -c -I/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/include -I/usr/opt/pmc/python/python-3.5.2.x86_64/include/python3.5m  -o /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/build/obj/x86_64/gpi_logging.o gpi_logging.c
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DIUS -Wstrict-prototypes -Waggregate-return -DFILTER -DIUS -shared -Xlinker -L/usr/opt/pmc/python/python-3.5.2.x86_64/lib -o /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/build/libs/x86_64/libgpilog.so /proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/build/obj/x86_64/gpi_logging.o   -lpthread -ldl -luti [... cut ...]
  File "/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/cocotb/__init__.py", line 35, in <module>
    import logging
  File "/usr/lib64/python2.6/logging/__init__.py", line 34, in <module>
    import sys, os, types, time, string, cStringIO, traceback
ImportError: /usr/lib64/python2.6/lib-dynload/timemodule.so: undefined symbol: PyExc_ValueError
Failed to load "cocotb"
ncsim: *W,RNQUIE: Simulation is complete.
ncsim> exit
Simulation time at exit is: 0 FS
make[1]: Leaving directory `/proj/digig5_1/devel/wakeford/coreotn_g5_tb/trunk/tbc/ip/scripts/cocotb/examples/endian_swapper/tests'
```

awakeford, 2018-05-15:
Note that the final error is:

ImportError: /usr/lib64/python2.6/lib-dynload/timemodule.so: undefined symbol: PyExc_ValueError

Which seems to indicate that in some way the wrong version of python is being picked up. i.e.. 2,6 instead of 3.5

mciepluc, 2018-05-15:
@awakeford 
Your problem is not related to the IUS itself. For some reason (probably the visibility of the PATH and other system variables across scripts calls) at some stage your PYTHON home is switched to 2.6 what causes the error.

I have 2 hints:
1. set PYTHONHOME in your Makefile to your preferred Python version e.g.:
```PYTHONHOME=/usr/local/anaconda3/```
```export PYTHONHOME```

2. Modify your PATH variable such way that preferred PYTHON executables are at the first place e.g.
```export PATH=/usr/local/anaconda3/bin/:$PATH```

awakeford, 2018-05-15:
Thanks @mciepluc 

That did work in that it changed some key parts of the build such that "-lpython2.6" replaced with "-lpython3.5m", which seems good.

Unfortunately on the first of these commands it not errors with:

/usr/bin/ld: /usr/opt/pmc/python/python-3.5.2.x86_64/lib/libpython3.5m.a(abstract.o): relocation R_X86_64_32S against `_PyObject_NextNotImplemented' can not be used when making a shared object; recompile with -fPIC
/usr/opt/pmc/python/python-3.5.2.x86_64/lib/libpython3.5m.a: could not read symbols: Bad value
collect2: ld returned 1 exit status

Which I think means that when abstract.o was compiled it was not position independent (no - fPIC). Seem like a problem with my Python install which I may not easily be able to fix.

stuarthodgson, 2018-07-29:
I think you do not have a shared library version of python 3.x (.so) only three archive (.a). You may need to create a symlink so another file, I'd suggest looking at how to ensure you have a full shared library python installing on your system and consult it's documentation.

stuarthodgson, 2018-07-29:
Please refresh open if you have issue with cocotb itself

closed 2018-07-29 by stuarthodgson