cocotb/cocotb #616: How to integrate a DPI-C code in cocotb?

tsengr0916, 2018-05-17:
I have a created a piece of C code that can pass the SV class object thru DPI-C, to Python. The code works fine without COCOTB. However, I have the problem to integrate this C code into COCOTB environment. Any helps/suggestions will be highly appreciated. 

The C code basically receives the string arguments from SV, and then calls another python function to process the arguments:

#include <Python.h>
#include <string.h>
#include <stdio.h>
#include "dpiheader.h"
#include "vpi_user.h"

int c_call_py(int argc, const svOpenArrayHandle argv)
{

PyObject *pName, *pModule, *pFunc;
PyObject *pArgs, *pValue;
char* c_argv[argc];

for (int i=0; i<argc; i++) {
  char *str;
  str = *(char **)svGetArrElemPtr1(argv,i);
  c_argv[i] = str;
  vpi_printf("C argv[%0d]=%s\n", i, c_argv[i]);      
}

if (argc < 2) {
   fprintf(stderr,"Usage: c_call_py pythonfile funcname [args]\n");
   return 1;
}

Py_Initialize();

// Error checking of pName left out
pName = PyUnicode_FromString(c_argv[0]);
  
pModule = PyImport_Import(pName);
Py_DECREF(pName);

if (pModule != NULL) {
    // pFunc is a new reference 
    pFunc = PyObject_GetAttrString(pModule, c_argv[1]);

    if (pFunc && PyCallable_Check(pFunc)) {
        pArgs = PyTuple_New(argc - 2);
        
        printf("C num args is %d\n",argc); 
                    
        for (int i = 0; i < argc - 2; i++) {
          pValue = PyUnicode_FromString(c_argv[i+2]);
          //vpi_printf("i=%0d\n", i);
          if (!pValue) {
            Py_DECREF(pArgs);
            Py_DECREF(pModule);
            fprintf(stderr, "Cannot convert argument\n");
            return 1;
          }
        
          // pValue reference stolen here:
          PyTuple_SetItem(pArgs, i, pValue);            
        }
        
        // call python function with args
        pValue = **PyObject_CallObject(pFunc, pArgs);**
        
        Py_DECREF(pArgs);
        if (pValue != NULL) {
            //vpi_printf("Result of call: %ld\n", PyLong_AsLong(pValue));
            Py_DECREF(pValue);
        }
        else {
            Py_DECREF(pFunc);
            Py_DECREF(pModule);
            PyErr_Print();
            fprintf(stderr,"Call failed\n");
            return 1;
        }
    }
    else {
        if (PyErr_Occurred())
            PyErr_Print();
        fprintf(stderr, "Cannot find function \"%s\"\n", c_argv[1]);
    }
    Py_XDECREF(pFunc);
    Py_DECREF(pModule);
}
else {
    PyErr_Print();
    fprintf(stderr, "Failed to load \"%s\"\n", c_argv[0]);
    return 1;
}
Py_Finalize();
return 0;
}

I compiled and linked the above C code to c_call_py.so file. Then in my Modelsim command, I did this:
vsim -sv_root /path/to/the/so/file/dir -sv_lib -sv_lib c_call_py ...

In my SV environment, I have this:
import "DPI-C" context function int c_call_py(int argc, input string argv[]);

And I could build an array and to call the C code from the SV code. This works fine if COCOTB is not involved. If I call the C code from SV within the COCOTB environment, it crashes and says this:

Fatal: (SIGSEGV) Bad handle or reference.
Time: 200 ns Iteration: 3 Process: /env_pkg::init_seq::body/#FORK#101(#ublk#207382391#101)_7fefe43402f File: /prj/tbs/5g/sandiego/hw/main/users/hctseng/dev5g/rxf/blocks/rx_front/verif/sim/env/src/tb_top_if.sv
Fatal: (vsim-4) ****** Memory allocation failure. *****
Attempting to allocate 140737438350376 bytes

Is there a better way to compile and link this C code along with other COCOTB C functions? Please advise. Thank you very much,

Richard

Nic30, 2019-01-10:
@tsengr0916 Maybe initialize Python interpret in [__attribute__ ((__constructor__))](https://stackoverflow.com/questions/24520305/ios-c-the-use-of-attribute-constructor-in-static-framework). 

But there are other problems and I expect that you have found out and this issue is already solved.

tsengr0916, 2019-01-10:
Yes, I have resolved lots of issues, and still more to overcome…

Richard

closed 2020-11-25 by github-actions[bot]