cocotb/cocotb #617: aldecpli.dll: file not recognized: File format not recognized

jhogans, 2018-05-21:
I am having an issue running cocotb with RivieraPRO.  I have contacted Aldec support who says they have several people in the office who can run it without a problem, while several others have the same issue I have.  They (naturally) suggest it is a cocotb issue.  I have tried running this with both the latest cocotb, as well as github.com/potentialventures/cocotb/tree/6c4320ee732b2af11faee17de57b9f40a4a5662c

When I run make, I get an error message:

c:/Aldec/Riviera-PRO-2018.02-x64/bin/aldecpli.dll: file not recognized: File format not recognized

I've tried a couple of different versions of Riviera.  I am using Python 2.7.9, and MinGW on a Windows 7 pro machine.  Any help or suggestions would be appreciated. Following is the full report.

$ make ALDEC_BIN_DIR=c:/Aldec/Riviera-PRO-2018.02-x64/bin SIM=aldec
make results.xml
make[1]: Entering directory `/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/examples/endian_swapper/tes
ts'
mkdir -p sim_build
echo "alib work" > sim_build/runsim.tcl
echo "set worklib work" >> sim_build/runsim.tcl
echo "alog +define+COCOTB_SIM -dbg -pli libgpi e:/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/examples/
endian_swapper/tests/../hdl/endian_swapper.sv" >> sim_build/runsim.tcl
echo "asim  +access +w -interceptcoutput -O2 -dbg -pli libvpi endian_swapper_sv" >> sim_build/runsim.tcl
echo "run -all" >> sim_build/runsim.tcl
echo "endsim" >> sim_build/runsim.tcl
mkdir -p /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/obj/i686
mkdir -p /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/libs/i686
make -C /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/utils SIM=aldec
make[2]: Entering directory `/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/utils'
gcc -m32 -g -DDEBUG -shared -DALDEC -c -I/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/include -Ic:\Py
thon27.9\include  -o /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/obj/i686/cocotb_utils.o cocot
b_utils.c
gcc -m32 -g -DDEBUG -shared -DALDEC -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Python27.9/libs -o /e/work/PersonalWork/cocotb/cocot
b-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/libs/i686/libcocotbutils.dll /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11fa
ee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/obj/i686/cocotb_utils.o
make[2]: Leaving directory `/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/utils'
make -C /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/gpi_log SIM=aldec
make[2]: Entering directory `/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/gpi_log'
gcc -m32 -g -DDEBUG -shared -DFILTER -DALDEC -c -I/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/includ
e -Ic:\Python27.9\include  -o /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/obj/i686/gpi_logging
.o gpi_logging.c
gcc -m32 -g -DDEBUG -shared -DFILTER -DALDEC -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Python27.9/libs -o /e/work/PersonalWork/coc
otb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/libs/i686/libgpilog.dll /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af
11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/obj/i686/gpi_logging.o   -lPython27
make[2]: Leaving directory `/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/gpi_log'
make -C /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/embed SIM=aldec
make[2]: Entering directory `/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/embed'
gcc -m32 -g -DDEBUG -shared -DPYTHON_SO_LIB=Python27.dll -DALDEC -c -I/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57
b9f40a4a5662c/include -Ic:\Python27.9\include  -o /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/
obj/i686/gpi_embed.o gpi_embed.c
gpi_embed.c: In function 'embed_init_python':
gpi_embed.c:108:81: warning: implicit declaration of function 'getpid' [-Wimplicit-function-declaration]
         fprintf(stderr, "Waiting for %lu seconds - Attach to %d\n", sleep_time, getpid());
                                                                                 ^~~~~~
gcc -m32 -g -DDEBUG -shared -DPYTHON_SO_LIB=Python27.dll -DALDEC -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Python27.9/libs -o /e/w
ork/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/libs/i686/libcocotb.dll /e/work/PersonalWork/cocotb/co
cotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/obj/i686/gpi_embed.o   -lPython27 -lgpilog -lcocotbutils -L/e/work/PersonalWork/
cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/libs/i686
make[2]: Leaving directory `/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/embed'
make -C /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/gpi EXTRA_LIBS=-laldecpli EXTRA_LIBDIRS=-Lc:
/Aldec/Riviera-PRO-2018.02-x64/bin SIM=aldec
make[2]: Entering directory `/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/gpi'
g++ -m32 -g -DDEBUG -shared -DVPI_CHECKING -DLIB_EXT=dll -DSINGLETON_HANDLES -DALDEC -c -I/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320e
e732b2af11faee17de57b9f40a4a5662c/include -Ic:\Python27.9\include  -o /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57
b9f40a4a5662c/build/obj/i686/GpiCbHdl.o GpiCbHdl.cpp
g++ -m32 -g -DDEBUG -shared -DVPI_CHECKING -DLIB_EXT=dll -DSINGLETON_HANDLES -DALDEC -c -I/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320e
e732b2af11faee17de57b9f40a4a5662c/include -Ic:\Python27.9\include  -o /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57
b9f40a4a5662c/build/obj/i686/GpiCommon.o GpiCommon.cpp
gcc -m32 -g -DDEBUG -shared -DALDEC -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Python27.9/libs -o /e/work/PersonalWork/cocotb/cocot
b-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/libs/i686/libgpi.dll  /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de5
7b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/obj/i686/GpiCbHdl.o /e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee
732b2af11faee17de57b9f40a4a5662c/build/obj/i686/GpiCommon.o  -lcocotbutils -lgpilog -lcocotb -lstdc++ -L/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/
cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/libs/i686
[... cut ...]
make[2]: Leaving directory `/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/lib/vhpi'
make[1]: *** [/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/build/libs/i686/libvhpi.dll] Error 2
make[1]: Leaving directory `/e/work/PersonalWork/cocotb/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/cocotb-6c4320ee732b2af11faee17de57b9f40a4a5662c/examples/endian_swapper/test
s'
make: *** [sim] Error 2

stuarthodgson, 2018-05-21:
```
c:/Aldec/Riviera-PRO-2018.02-x64/bin/aldecpli.dl
```
This suggests that you are using 64bit riviera, but
```
gcc -m32 -g -DDEBUG -shared -DALDE ...
```
Suggests you are compiling as 32bit, do you have 64bit Mingw installed and 64 bit python?

stuarthodgson, 2018-07-29:
Seems like python liking errors due to an inconsistent windows environment.

closed 2018-07-29 by stuarthodgson