cocotb/cocotb #626: Cocotb initialisation failed , How to debug it?

kirknodeng, 2018-07-17:
I use IUS 15.20.013 simulator.

When I start cocotb, simulator output below:

     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
...
...
...
...
ncsim> source /sw/Cadence/Incisive/Incisive15.20.013/tools/inca/files/ncsimrc
ncsim> run
Cocotb initialisation failed - exiting
Simulation complete via $finish(1) at time 0 FS + 7
ncsim> exit

So, I want to find problem. I found that string in gpi_embed.c. 
How can I debug this case? 

Sometimes cocotb is work, but, almost initialization failed.

kirknodeng, 2018-07-17:
I updated new version. 
It has no problem. 

I close this issue

kirknodeng, 2018-07-17:
I trying to use new cocotb. 
but, this issue is not fixed. 

simulation message was changed, 

ncsim> source /sw/Cadence/Incisive/Incisive15.20.013/tools/inca/files/ncsimrc
ncsim> run
Cocotb initialisation failed - exiting
    0.00ns ERROR    cocotb.scheduler                            __init__.py:177  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely

stuarthodgson, 2018-07-29:
There is nothing here that we can help you with as there is no information provided. I suspect b you have an error in your testbench. I've done some merges recently that add summer more logging to worry cases people initially hit.

closed 2018-07-29 by stuarthodgson