cocotb/cocotb #627: Cocotb website frozen in version 0.4

Martoni, 2018-07-26:
Is it normal that cocotb website describe version 0.4 ?
http://potential.ventures/cocotb/

It's a really old revision no ?

eric-wieser, 2018-07-30:
I think http://cocotb.readthedocs.io/en/latest/ might be up to date.

@stuarthodgson, can you set the (currently absent) url in the github UI to point to http://cocotb.readthedocs.io/en/latest/?

eric-wieser, 2018-08-03:
I think the cocotb website comes from the `gh-pages` branch, which is similarly out of date

Martoni, 2018-08-16:
The problem is that if we google/duckduckgo cocotb this obsolete link appear.

stuarthodgson, 2018-10-29:
Mm, ok, not quite sure where that is even hosted. We will track it down.

eric-wieser, 2018-10-29:
It's hosted by github, on the `gh-pages` branch.

Martoni, 2018-11-06:
It could be a good idea to delete these gh-pages. Because with simple search we have the impression that cocotb is old unmaintained project. Which is false !

stuarthodgson, 2018-11-06:
If I search for cocotb it points to the 1.0.0 pages so I'm now unsure what the problem even is here.

eric-wieser, 2018-11-06:
Fourth link on google:

[image]

Take it down with `git push -d origin gh-pages` (unless you want to back that branch up first

eric-wieser, 2018-11-06:
Also, ping on [this comment](https://github.com/potentialventures/cocotb/issues/627#issuecomment-408726884)

Martoni, 2018-11-06:
For me, duckduckgo is first link : 
[image]

closed 2018-11-15 by themperek