cocotb/cocotb #631: Easy way to use different Python install

juliusbaxter, 2018-07-29:
I'm working on a system with an ancient Python version installed in /usr/bin/python, which cocotb appears to pick up when running.

I've got Python 2.7 and 3.4 installed on the system too but in different locations and with different names; python27 and python3 respectively.

I would like to try using them but I can't find any way to change the install of Python cocotb picks up. I've tried overriding PYTHON_BIN, and aliasing python to another one of the installs to no avail.

Does anyone know of an easy way to get cocotb to use one of the others?

Thanks

stuarthodgson, 2018-07-29:
It uses python-config and python so purley depends on the PATH variable. I used to use pyemv to manage and gets with different versions back in the day. Stu

elgorwi, 2018-07-29:
Currently running anaconda python 2.7 rather than the built-in python on centos 7.3. When running a test I do following: 
>make PATH=/path_to_anaconda/bin:$PATH PYTHONHOME=/path_to_anaconda

Had to make a change to Makefile.pylib.Linux to ensure we pick up python-config from anaconda and not /usr/bin/python2.7-config. (moved up line 64-65 before line 60)

juliusbaxter, 2018-07-30:
Thanks guys, both were helpful.

I managed to get it going for the time being with Python 2.7 by doing something awful, but similar to what pyenv does:

```
mkdir $HOME/python-shims
ln -s /usr/bin/python2.7 $HOME/python-shims/python
ln -s /usr/bin/python2.7-config $HOME/python-shims/python-config
export PATH=$HOME/python-shims:$PATH
```

I didn't need to set PYTHONHOME when I did it this way.

My 3.4 install didn't want to play when running with ``PYTHONHOME=/tools/python3/install/3.4.5``, I got:
```
Failed to find python lib
Unable to open lib libpython3.4m.a (/tools/python3/install/3.4.5/lib/libpython3.4m.a: invalid ELF header)
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:114  in embed_init_python               Did not detect virtual environment. Using system-wide Python interpreter.
...
Failed to load "cocotb"
ncsim: *W,RNQUIE: Simulation is complete.
```

Not sure what's going on there, python3 definitely runs on its own.

Anyway I got it going, and it seems like this problem is outside of the scope of cocotb so I'll close this.

Thanks again.

closed 2018-07-30 by juliusbaxter

nevsan, 2018-07-30:
The best way to handle this is through [pyenv](https://github.com/pyenv/pyenv), which got a lot easier now that https://github.com/potentialventures/cocotb/pull/615 is merged.