cocotb/cocotb #637: coroutines that return before their first yield cause the simulator to shutdown

eric-wieser, 2018-08-01:
Repro:
```
@cocotb.test()
def test_func_empty(dut):
    """ Test that a function can complete before the first yield """
    @cocotb.coroutine
    def func_empty():
        return
        yield  # needed to make this a coroutine
    yield func_empty()
```

eric-wieser, 2018-10-28:
A less contrived example is:

```python
@cocotb.test()
def test_func_empty(dut):
    """ Test that a function can complete before the first yield """
    @cocotb.coroutine
    def wait_if_even(x):
        print("This line runs")
        if x % 2 == 0:
            yield Timer(x)
    yield wait_if_even(2)
    print("This line also runs")
    yield wait_if_even(3)
    print("This line is never reached")
```

eric-wieser, 2019-07-17:
Note that this was auto-closed, but by a commit that said "mostly fix".

garmin-mjames, 2020-04-29:
@eric-wieser looks like #1705 does fix the rest of this.
```python3
@cocotb.test()
def test_immediate_test(dut):
    """
    Test that tests can return immediately
    """
    return
    yield
```
Passes 👍

eric-wieser, 2020-04-29:
This issue doesn't track the problem very well any more. If you looks through the tests, there are a few that still reference 637, and fail without the line the comment describes. I think they concern `function` and `external`.

eric-wieser, 2020-04-30:
One relevant test is this one:

https://github.com/cocotb/cocotb/blob/5ab94f03c179c839899badb741e7113c30a34923/tests/test_cases/test_external/test_external.py#L207-L222

ktbarrett, 2020-05-06:
@eric-wieser What's the error here? Declaring the external function before starting the clock?

eric-wieser, 2020-05-06:
The failure mode is starting an `external` at the beginning of a test, such that the first wakeup is the external finishing, and not some other trigger.

garmin-mjames, 2020-05-06:
If I remove all of the `clk_gen = cocotb.fork(Clock(dut.clk, 100).start())` lines and run the `test_external` module on #1705, everything passes.

eric-wieser, 2020-05-06:
Milestoning as 1.4 so we remember to remove all those lines and test before releasing.

garmin-mjames, 2020-05-06:
`test_ext_exit_error` passes without error when I don't skip it, so we get a FAIL on that. It needs reworking to get the desired result.

eric-wieser, 2020-05-07:
It looks to me like `test_ext_exit_error` is just nonsense - I'd remove it as part of the cleanup.

PR #1776 (2020-05-07): gh-637 cleanup
Closes #637.
Follow on from #1705.

Now that tests are started with `react`, it's no longer necessary to `await` a `GPITrigger` to ensure the correct `scheduler` behavior.

While I was there I cleaned up `test_external`.

I tested IPython on Windows with Active-HDL and on WSL1 Ubuntu with iverilog.
closed 2020-05-07 by eric-wieser