cocotb/cocotb #639: solved/fyi: cadence IUS libstdc++.so causes libvpi.so not to load

ashmanskas, 2018-08-02:
I recently encountered (and finally tracked down, I believe) an issue that broke my use of Cadence IUS with Cocotb on Ubuntu 16.04 and 18.04.  If I understand correctly, the issue is that Cadence includes its own libstdc++.so.6.0.19 in INCISIVE152-lnx/tools.lnx86/lib/, while Cocotb's Makefile produces a libvpi.so that depends on /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25.  The result is that libvpi.so silently fails to load (though this can be debugged somewhat by changing EXTRA_ARGS from "-plinowarn" to "-pliverbose" in Makefile.ius):

====
% SIM=ius make
...
	Primary VPI symbols.
		vlog_startup_routines:  builtin default
...
	Processing the vlog_startup array:
ncsim> source /cad/INCISIVE152-lnx/tools/inca/files/ncsimrc
ncsim> run
ncsim: *W,RNQUIE: Simulation is complete.
ncsim> exit
====

whereas if I preload the system's own libstdc++.so, then all Cocotb tests run properly:

====
% LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 SIM=ius make
...
	Primary VPI symbols.
		vlog_startup_routines:  /raid/ashmansk/code/cocotb/build/libs/x86_64/libvpi.so
...
	Processing the vlog_startup array:
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:114  in embed_init_python               Did not detect virtual environment. Using system-wide Python interpreter.
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
...
ncsim> source /cad/INCISIVE152-lnx/tools/inca/files/ncsimrc
ncsim> run
    0.00ns INFO     cocotb.gpi                                  gpi_embed.c:281  in embed_sim_init                  Running on ncsim(64) version 15.20-s046
    0.00ns INFO     cocotb.gpi                                  gpi_embed.c:282  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
    0.00ns INFO     cocotb                                      __init__.py:118  in _initialise_testbench           Running tests with Cocotb v1.0.1 from /raid/ashmansk/code/cocotb
    0.00ns INFO     cocotb                                      __init__.py:134  in _initialise_testbench           Seeding Python random module with 1533228538
    0.00ns INFO     cocotb.regression                         regression.py:188  in _initialise                     Found test test_endian_swapper.run_test_001
    0.00ns INFO     cocotb.regression                         regression.py:188  in _initialise                     Found test test_endian_swapper.run_test_002
    0.00ns INFO     cocotb.regression                         regression.py:188  in _initialise                     Found test test_endian_swapper.run_test_003
    0.00ns INFO     cocotb.regression                         regression.py:188  in _initialise                     Found test test_endian_swapper.run_test_004
...
ndian_swapper.run_test_032     PASS      1875001.00          0.12   15290662.65  **
                                                                                                                       ** test_endian_swapper.wavedrom_test    PASS        40001.00          0.00   8525221.26  **
                                                                                                                       *******************************************************************************************
                                                                                                                       
31980033.00ns INFO     cocotb.regression                         regression.py:390  in _log_sim_summary                *************************************************************************************
                                                                                                                       **                                 ERRORS : 0                                      **
                                                                                                                       *************************************************************************************
                                                                                                                       **                               SIM TIME : 31980033.00 NS                         **
                                                                                                                       **                              REAL TIME : 2.74 S                                 **
                                                                                                                       **                        SIM / REAL TIME : 11684004.18 NS/S                       **
                                                                                                                       *************************************************************************************
                                                                                                                       
31980033.00ns INFO     cocotb.regression                         regression.py:219  in tear_down                       Shutting down...

	PLIVERBOSE API stats
		Handle counts
		   PLI/VPI handles:  55
Simulation complete via $finish(1) at time 31980033 NS + 0
ncsim> exit
--- ncdbg_exit() has been called
====

The way that I finally narrowed down the problem to libstdc++ was by running the following, in the examples/adder/tests directory:

====
% LD_LIBRARY_PATH=../../../build/libs/x86_64 LD_PRELOAD=../../../build/libs/x86_64/libvpi.so  irun -64 -nclibdirpath sim_build  -pliverbose  -sv -v93 -loadvpi /raid/home/ashmansk/petdaq/build/libs/x86_64/libvpi.so +access+rwc  /raid/home/ashmansk/petdaq/examples/adder/tests/../hdl/adder.v
ERROR: ld.so: object '../../../build/libs/x86_64/libvpi.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
irun(64): 15.20-s046: (c) Copyright 1995-2018 Cadence Design Systems, Inc.
ncsim: /tape/cad/INCISIVE152-lnx/tools.lnx86/inca/bin/64bit/../../../lib/64bit/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by ../../../build/libs/x86_64/libvpi.so)
ncsim: /tape/cad/INCISIVE152-lnx/tools.lnx86/inca/bin/64bit/../../../lib/64bit/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by ../../../build/libs/x86_64/libvpi.so)
ncsim: /tape/cad/INCISIVE152-lnx/tools.lnx86/inca/bin/64bit/../../../lib/64bit/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by ../../../build/libs/x86_64/libgpi.so)
ncsim: /tape/cad/INCISIVE152-lnx/tools.lnx86/inca/bin/64bit/../../../lib/64bit/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by ../../../build/libs/x86_64/libgpi.so)
====

That was the crucial hint that libstdc++.so.6 was being picked up from the Cadence install area rather than from /usr/lib/x86_64-linux-gnu.

So for now I am simply pre-loading /usr/lib/x86_64-linux-gnu/libstdc++.so.6 using LD_PRELOAD when I invoke Cadence IUS via Cocotb.

I post this here in case it is of use to someone else using Cocotb with Cadence INCISIVE.  I would also be happy to hear if there is a better solution that I overlooked.

-Bill

lukedarnell, 2018-09-26:
We could detect what distro we're running on and tweak LD_PRELOAD when invoking irun.
Not sure if we want to start polluting the makefiles with distro specific hacks.
Will discuss on the mailing list and update this issue.

ktbarrett, 2020-09-10:
There is [documentation](https://docs.cocotb.org/en/stable/troubleshooting.html#glibcxx-3-4-xx-not-found) on this problem.

closed 2020-09-10 by ktbarrett