cocotb/cocotb #644: ImportError with import hashlib.py

flamehj, 2018-08-13:
I get this error when running the examples:
`Traceback (most recent call last):`
` File "/home/d.baburin/work/cocotb/cocotb/__init__.py", line 37, in <module>`
`   import random`
` File "/usr/lib64/python2.7/random.py", line 49, in <module>`
`   import hashlib as _hashlib`
` File "/usr/lib64/python2.7/hashlib.py", line 116, in <module>`
`   import _hashlib`
`ImportError: /usr/lib64/libssl.so.10: symbol private_ossl_minimum_dh_bits, version libcrypto.so.10 not defined in file libcrypto.so.10 with link time reference`
`Failed to load "cocotb"`
`
I have Centos 7 and Cadence Xcelium simulator (new version of Cadence IUS). 

As far as I understand it requires earlier version for openssl. My version 1.0.2k and in /lib64/ libssl.so.10 and libcrypto.so.10 symlinks links on 1.0.2k version files/ When I change simlinks on previous version - 0.9.8e, I got next error( but simulation was started):
`Traceback (most recent call last):`
` File "/usr/libexec/abrt-action-ureport", line 11, in <module>`
`   import augeas`
` File "/usr/lib/python2.7/site-packages/augeas.py", line 43, in <module>`
`   import ctypes.util`
` File "/usr/lib64/python2.7/ctypes/util.py", line 89, in <module>`
`   import re, tempfile, errno`
` File "/usr/lib64/python2.7/tempfile.py", line 35, in <module>`
`   from random import Random as _Random`
` File "/usr/lib64/python2.7/random.py", line 49, in <module>`
`   import hashlib as _hashlib`
` File "/usr/lib64/python2.7/hashlib.py", line 116, in <module>`
`   import _hashlib`
`ImportError: /usr/lib64/python2.7/lib-dynload/_hashlib.so: symbol HMAC_CTX_copy, version libcrypto.so.10 not defined in file libcrypto.so.10 with link time reference`

jrpetrus, 2018-09-12:
I upgraded to Xcelium and get this same error from `xrun` when loading my DPI-C code that links in Octave 4.0.3 routines I wrote.
```
xmsim: *W,LIBRUN: Could not load the dynamic library:  ./xcelium.d/tb_default_opt.lnx8664.18.03.d/librunpost
	System ERROR:  /usr/lib64/libssl.so.10: symbol private_ossl_minimum_dh_bits, version libcrypto.so.10 not defined in file libcrypto.so.10 with link time reference.
```
I'm also on CentOS 7 and running the latest Xcelium (18.03-s011).

Cocotb works fine with Xcelium for me, but I use Python 3.6.3.

jrpetrus, 2018-09-13:
Thanks, @stuarthodgson! That fixed it! I only searched for `libssl*` and assumed that wasn't the problem.

@flamehj - I renamed the library files and changed the symlinks in the XCELIUM directories:
```
./tools.lnx86/lib/libcrypto.so.1.0.1e --> renamed to not_used_libcrypto.so.1.0.1e
./tools.lnx86/lib/libcrypto.so.10 --> changed symlink to /usr/lib/libcrypto.so.10
./tools.lnx86/lib/64bit/libcrypto.so.1.0.1e  --> renamed to not_used_libcrypto.so.1.0.1e
./tools.lnx86/lib/64bit/libcrypto.so.10 --> changed symlink to /usr/lib64/libcrypto.so.10
```

flamehj, 2018-09-18:
Thanks, @stuarthodgson, @jrpetrus! It really work!

closed 2018-09-18 by flamehj

martijnbastiaan, 2021-07-07:
Alternatively, if you don't want or cannot change your Xcelium distribution you can patch these lines:

https://github.com/cocotb/cocotb/blob/4946a08d744f2b96ca0e85d62efba3203710da10/cocotb/share/makefiles/simulators/Makefile.xcelium#L110-L112

Splicing an extra line `LD_PRELOAD="/lib64/libcrypto.so.10 /lib64/libssl.so.10" \` in between such that it reads:

```makefile
	MODULE=$(MODULE) TESTCASE=$(TESTCASE) TOPLEVEL=$(TOPLEVEL) GPI_EXTRA=$(GPI_EXTRA) TOPLEVEL_LANG=$(TOPLEVEL_LANG) \
	LD_PRELOAD="/lib64/libcrypto.so.10 /lib64/libssl.so.10" \
	$(CMD) -timescale $(COCOTB_HDL_TIMEUNIT)/$(COCOTB_HDL_TIMEPRECISION) \
	$(EXTRA_ARGS) $(GPI_ARGS) $(INCDIRS) -access +rwc -createdebugdb $(MAKE_LIB) $(HDL_SOURCES) $(PLUSARGS) 2>&1 | tee $(SIM_BUILD)/sim.log

```

seems to work for me.