cocotb/cocotb #648: The problem in Windows7

cnshijin, 2018-08-20:
Hi,

I using the newest version, and installed it following by  [Install cocotb on Windows 10](https://dornerworks.com/blog/install-cocotb-windows-10-boost-fpga-verification-productivity), but my actual OS is Window7-64bit. When I ran the adder's example in cocotb, it didn't work. Here is the log:

```
make -C /e/a_project/cocotb/lib/utils SIM=modelsim
make[2]: 进入目录“/e/a_project/cocotb/lib/utils”
make -C /e/a_project/cocotb/lib/utils SIM=modelsim
make[2]: 进入目录“/e/a_project/cocotb/lib/utils”
gcc -g -DDEBUG -shared -DMS_WIN64 -DMODELSIM -shared -Wl,-no-undefined -Wl,-enab                                                                                               le-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/d/Anaconda3/libs -o /e/a_                                                                                               project/cocotb/build/libs/x86_64/libcocotbutils.dll /e/a_project/cocotb/build/ob                                                                                               j/x86_64/cocotb_utils.o
E:/a_project/cocotb/build/obj/x86_64/cocotb_utils.o:cocotb_utils.c:(.rdata$.refp                                                                                               tr.context[.refptr.context]+0x0): undefined reference to `context'
collect2.exe: error: ld returned 1 exit status
```

Thank you for replying.

elgorwi, 2018-08-20:
This problem has been reported in issue #585. You can get around it by merging pull-request #620.

stuarthodgson, 2018-08-20:
One merged this to master now.

closed 2018-08-20 by stuarthodgson