cocotb/cocotb #650: Restarting Clock() objects in each test

juliusbaxter, 2018-08-21:
I have a setup function which is run for each test, and initializes some things in the testbench as well as creating the clocks.

When I do a run of all tests in my module, the first one goes without an issue but every subsequent one crashes when trying to kill the clock and restart it.

My code looks like this:

```
clk_sdio = None
clk_sdio_gen = None

def test_setup(dut):

    # Initialize SDIO clock to as fast as possible, so 50MHz for SDIO 2.0
    set_clocks(dut,sdio_clk_period=20.0)
   
    # and other testbench init stuff ...

def set_clocks(dut,sdio_clk_period=20.0,sdio_clk_units="ns"):
    """
    Stop and restart the clock
    """
    global clk_sdio
    global clk_sdio_gen
    if clk_sdio_gen:
        clk_sdio_gen.kill()
    clk_sdio = Clock(dut.sdio_clk, sdio_clk_period, units=sdio_clk_units)
    clk_sdio_gen = cocotb.fork(clk_sdio.start())
    dut.log.info("SDIO clock period set to %s%s" %(sdio_clk_period,sdio_clk_units))
   
            
@cocotb.test(skip = False)
def test_1(dut):
    """
    Description:
        Initialize the SDIO
    """
    tb,drv = test_setup(dut)
    # ... test goes on ...

@cocotb.test(skip = False)
def test_2(dut):
    """
    Description:
        Send some SDIO data commands
    """
    tb,drv = test_setup(dut)
    # ... test goes on ...

```

As you can see, I'm storing the forked clock thread in the clk_sdio_gen global variable, which I guess isn't a big deal.

There's no issue with re-calling this function in the same test (if it runs first); I'm able to call set_clocks() again with a new frequency and that works. It's just an issue from one test to the next. The crash message I get is:

```
Send raised exception: 'RunningCoroutine' object has no attribute '_join'
                        File "/work/cocotb/cocotb/decorators.py", line 213, in send
                          return self._coro.send(value)
                        File "/work/sdio-device/tb/sdio_device_tests.py", line 1162, in data_tests_4bit
                          tb,drv = test_setup(dut)
                        File "/work/sdio-device/tb/sdio_device_tests.py", line 1085, in test_setup
                          set_clocks(dut,sdio_clk_period=20.0)
                        File "/work/sdio-device/tb/sdio_device_tests.py", line 1116, in set_clocks
                          clk_sdio_gen.kill()
                        File "/work/cocotb/cocotb/decorators.py", line 150, in kill
                          cocotb.scheduler.unschedule(self)
                        File "/work/cocotb/cocotb/scheduler.py", line 453, in unschedule
                          if coro._join in self._trigger2coros:
```

So can anyone suggest what I should be doing here instead? I don't really understand the issue. Are all coroutines killed between tests? How do I test whether it's running or not, because the test I do is whether the object is not None.

juliusbaxter, 2018-08-22:
So if I do a dir(sdio_clk_gen) before and after the cocotb.fork() like so:

```
def set_clocks(dut,sdio_clk_period=20.0,sdio_clk_units="ns"):
    """
    Stop and restart the clock
    """
    global clk_sdio
    global clk_sdio_gen
    dut.log.info("before : %s, and dir: %s" %(type(clk_sdio_gen), dir(clk_sdio_gen)))
    if clk_sdio_gen:
        clk_sdio_gen.kill()
    clk_sdio = Clock(dut.sdio_clk, sdio_clk_period, units=sdio_clk_units)
    clk_sdio_gen = cocotb.fork(clk_sdio.start())
    dut.log.info("type of cocotb.fork returned item: %s, and dir: %s" %(type(clk_sdio_gen), dir(clk_sdio_gen)))
```
 
I get this the first time through, as expected:

```
    0.00ns INFO     before type of cocotb.fork returned item: <type 'NoneType'>, and dir: ['__class__', '__delattr__', '__doc__', '__format__', '__getattribute__', '__hash__', '__init__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__']
    0.00ns INFO     type of cocotb.fork returned item: <class 'cocotb.decorators.RunningCoroutine'>, and dir: ['__class__', '__delattr__', '__dict__', '__doc__', '__format__', '__getattribute__', '__hash__', '__init__', '__iter__', '__module__', '__name__', '__new__', '__nonzero__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_callbacks', '_coro', '_finished', '_finished_cb', '_join', '_parent', '_started', 'close', 'funcname', 'has_started', 'join', 'kill', 'log', 'module', 'retval', 'send', 'throw']
```

... then for the second test:

```
33590.00ns INFO     before : <class 'cocotb.decorators.RunningCoroutine'>, and dir: ['__class__', '__delattr__', '__dict__', '__doc__', '__format__', '__getattribute__', '__hash__', '__init__', '__iter__', '__module__', '__name__', '__new__', '__nonzero__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_callbacks', '_coro', '_finished', '_finished_cb', '_parent', '_started', 'close', 'funcname', 'has_started', 'join', 'kill', 'log', 'module', 'retval', 'send', 'throw']
33590.00ns ERROR    Send raised exception: 'RunningCoroutine' object has no attribute '_join'
                      File "/work/cocotb/cocotb/decorators.py", line 213, in send
                        return self._coro.send(value)
                      File "/work/sdio-device/tb/sdio_device_tests.py", line 1143, in init_test_rtl_host
                        tb,drv = test_setup(dut)
                      File "/work/sdio-device/tb/sdio_device_tests.py", line 1085, in test_setup
                        set_clocks(dut,sdio_clk_period=20.0)
                      File "/work/sdio-device/tb/sdio_device_tests.py", line 1117, in set_clocks
                        clk_sdio_gen.kill()
                      File "/work/cocotb/cocotb/decorators.py", line 150, in kill
                        cocotb.scheduler.unschedule(self)
                      File "/work/cocotb/cocotb/scheduler.py", line 453, in unschedule
                        if coro._join in self._trigger2coros:

```

There is indeed a missing _join attribute on the RunningCoroutine object clk_sdio_gen when the second test begins. The other interesting thing is that's the *only* one not there.

How can that be? ¯\\\_(ツ)\_/¯

pgfarley, 2018-11-13:
Storing sdio_clk_gen in global state is causing the RunningCoroutine to be unscheduled twice; once by the Scheduler at the end of the first test, and then a second time by set_clocks at the start of the second test. Scheduler.unschedule is not idempotent.

tl;dr

At the end of the first test Scheduler.finish_test() invokes Scheduler.cleanup()  which calls Scheduler.kill() on all waiting coros including clk_sdio_gen. 

Scheduler.unschedule deletes the _join attribute off the RunningCoroutine to aid garbage collection. I don't know enough about Python's GC to understand why this is necessary, but maybe assigning None would be equivalent?

https://github.com/potentialventures/cocotb/blob/bc1948357feb317c2618cf79f0f5774d6f089a88/cocotb/scheduler.py#L457

When the second test invokes set_clocks it attempts to kill the already unscheduled clk_sdio_gen. It's arguably a good idea for this to throw an exception, although probably something more intentional then the no attribute found. 

Three suggested ways to make this nicer:
1) Raise a more descriptive exception when an unscheduled coroutine is unscheduled
2) Make unschedule idempotent. Means at least changing the del _join solution (maybe with assignment to None). There may be other issues with this lurking.
3) Add support for RunningCoroutines that can survive individual tests 

I'm happy to submit a patch for 1) which is maybe enough to just close this. It's not obvious to me that 2) or 3) are worth the effort, or even desirable. I'd be willing to take a look at 2) if there's a strong opinion.

forensicgarlic, 2018-12-12:
good exception messages seem like a good first start.

imphil, 2018-12-16:
@juliusbaxter does the current master with #723 solve this issue for you?

juliusbaxter, 2019-01-30:
@imphil yes I just tested it and it works fine now! Thanks very much for the fix, great job all!

closed 2019-01-30 by juliusbaxter