cocotb/cocotb #651: (SIGSEGV) Bad pointer access in Windows7 64-bit with Modelsim 10.2c

cnshijin, 2018-08-22:
Hi,

When I tested the example of Endian Swapper, it didn't work. The log as the following:

'''
# Attempting stack trace sig 11
# Signal caught: signo [11]
# vsim_stacktrace.vstf written
# Current time Wed Aug 22 09:53:24 2018
# ModelSim Stack Trace
# Program = vsim
# Id = "10.2c"
# Version = "2013.07"
# Date = "Jul 19 2013"
# Platform = win64
# 0    0x000000006e785d21: '<unknown (@0x6e785d21)>'
# 1    0x000000006e783514: '<unknown (@0x6e783514)>'
# 2    0x0000000066b82ed3: '<unknown (@0x66b82ed3)>'
# 3    0x0000000063ea034d: '<unknown (@0x63ea034d)>'
# 4    0x0000000063e9fa76: '<unknown (@0x63e9fa76)>'
# 5    0x0000000063e7f1a0: '<unknown (@0x63e7f1a0)>'
# 6    0x0000000063e9f51e: '<unknown (@0x63e9f51e)>'
# 7    0x0000000063e9f3d0: '<unknown (@0x63e9f3d0)>'
# 8    0x0000000063e75103: '<unknown (@0x63e75103)>'
# 9    0x0000000063e714f7: '<unknown (@0x63e714f7)>'
# 10   0x0000000063eb9188: '<unknown (@0x63eb9188)>'
# 11   0x0000000063e9ef51: '<unknown (@0x63e9ef51)>'
# 12   0x0000000063ea0253: '<unknown (@0x63ea0253)>'
# 13   0x0000000063e9fa76: '<unknown (@0x63e9fa76)>'
# 14   0x0000000063e7f1a0: '<unknown (@0x63e7f1a0)>'
# 15   0x0000000063e9fc3b: '<unknown (@0x63e9fc3b)>'
# 16   0x0000000063e7f1a0: '<unknown (@0x63e7f1a0)>'
# 17   0x0000000063ea07af: '<unknown (@0x63ea07af)>'
# 18   0x0000000063e9f5e8: '<unknown (@0x63e9f5e8)>'
# 19   0x0000000063e9f3d0: '<unknown (@0x63e9f3d0)>'
# 20   0x0000000063e72a19: '<unknown (@0x63e72a19)>'
# 21   0x0000000066b816c1: '<unknown (@0x66b816c1)>'
# 22   0x000000006e781829: '<unknown (@0x6e781829)>'
# 23   0x0000000070f099a2: '<unknown (@0x70f099a2)>'
# 24   0x0000000070f03644: '<unknown (@0x70f03644)>'
# 25   0x0000000140234984: '<unknown (@0x140234984)>'
# 26   0x000000014001b8ea: '<unknown (@0x14001b8ea)>'
# 27   0x00000001400fd369: '<unknown (@0x1400fd369)>'
# 28   0x00000001404ce887: '<unknown (@0x1404ce887)>'
# End of Stack Trace

** Fatal: (SIGSEGV) Bad pointer access. Closing vsimk.
** Fatal: vsimk is exiting with code 211.
(Exit codes are defined in the ModelSim messages appendix
of the ModelSim User's Manual.)
# Attempting stack trace sig 11
# Signal caught: signo [11]
'''

The cocotb is the newest at master branch. Thank you!

eric-wieser, 2018-08-22:
Almost certainly due to `sizeof(long) < sizeof(void*)` on windows, and cocotb is dropping the top bits of pointers when converting them to python ints.

cnshijin, 2018-08-23:
And how to fix it? @eric-wieser

eric-wieser, 2018-08-23:
By making cocotb not use `PyObject* = PyLong_FromLong(void*)` or `void *x = PyLong_AsLong(PyObject*)` internally

cnshijin, 2018-08-24:
Thank you for your replying. I'm not familar with python.

cnshijin, 2018-08-24:
The error is the same in questasim_10.4e.

eric-wieser, 2018-08-24:
#654 may fix this

cnshijin, 2018-08-26:
The problem is fixed. Thank you for your excellent work @eric-wieser .

closed 2018-08-26 by cnshijin

eric-wieser, 2018-08-26:
@cnshijin: Fixed by using the unmerged patch from #654, or it just went away by itself?

cnshijin, 2018-08-26:
@eric-wieser Fixed by using the unmerged patch from #654.

eric-wieser, 2018-08-26:
You should comment on #654 that the patch works for you, to encourage that it be merged