cocotb/cocotb #653: cocotb fails to initialize in 32-bit Aldec since #629

jeroenvanstraten, 2018-08-22:
I'm getting [this error message](https://github.com/potentialventures/cocotb/files/2310091/simlog.txt) when running *any* cocotb testbench in the following environment:

 - 32-bit Riviera-PRO 2017.10.67.6735
 - 32-bit Python 3.6.5
 - VHDL/VHPI interface
 - Running on 64-bit CentOS (we're using the 32-bit simulator for licensing reasons)

This happens since [this commit](https://github.com/potentialventures/cocotb/pull/629/commits/56c4ec1428343b63caad02a9b576b4b5cf9b2403) in #629. @eric-wieser, you mentioned that you found more issues related to #629, do you expect that those will eventually fix this problem as well?

stuarthodgson, 2018-08-22:
Not sure they could, they would introduce the problem in more places.

eric-wieser, 2018-08-22:
I know the cause here.

To summarize #629 - I changed `Py_BuildValue("l", hdl);` to `PyLong_FromVoidPtr`.This has the following effect:

| C `void*` value | platform | old python value | new python value |
| --- | --- | --- | --- |
| `0x01234567_01234567` | Windows 64-bit |  `0x01234567` (bad bug) | `0x01234567_01234567` |
| `0x01234567_01234567` | Other 64-bit |  `0x01234567_01234567` | `0x01234567_01234567` |
| `0x01234567` | 32-bit |  `0x01234567` | `0x01234567` |
| `0x80000000` | 32-bit |  `-0x80000000` | `0x80000000` |

This last row is causing your problem, as elsewhere the python integer is loaded with `Py_ParseTupleArgs("l")`, where `l` is a _signed_ `long`. `0x80000000` doesn't fit in a signed long, so this fails.

The  fix here is to replace all these formats with `"O"`, and then call `PyLong_FromVoidPtr` on the result. 

Changing the format strings to `L` would s

eric-wieser, 2018-08-22:
Another fix would be:
```
int VoidPtrConverter(PyObject* o, void** data) {
   void *p = PyLong_AsVoidPtr(o);
   if (p == -1 && PyErr_Occurred()) {
       return 1;
   }
   *data = p;
   return 0;
}
```
PyArg_ParseTuple("O&", VoidPtrConverter, &hdl)
```

jeroenvanstraten, 2018-08-22:
Awesome, I can work with that. I'll try it out tomorrow and submit a PR when I get it to work again.

PR #654 (2018-08-26): Use PyLong_AsVoidPtr
Fixes #653, see issue for details. @eric-wieser, can you check if this works in your environment as well?
closed 2018-08-26 by stuarthodgson