cocotb/cocotb #656: BUG: gpi_set_signal_value_long ignores the top 32 bits on platforms where sizeof(long) != 4

eric-wieser, 2018-08-29:
All the current implementations of `GpiSignalObjHdl::set_signal_value(long)` end up casting to `uint32`, which loses bits on 64-bit non-windows.

Easy fix would be just to rename it to `gpi_set_signal_value_uint32`, and change the types of the parameters to `uint32`

elgorwi, 2018-08-30:
Handle.py ensures that only values that fit within 32-bit and HDL signals up to 32-bit are driven using set_signal_val_long. 
Fair point about changing this to uint32 though.
 
On another note, perhaps another method using hex strings (eg. set_signal_val_hex) could be introduced to drive larger signals more efficiently than set_signal_val_str.

stuarthodgson, 2018-08-31:
I don't think there is anything that needs fixing here, would just be code churn.

closed 2018-09-13 by stuarthodgson