cocotb/cocotb #664: COCOTB crashes on QuestaSim 10.7b

tsengr0916, 2018-09-16:
I am seeing cocotb crashes when upgrading QuestaSim simulator to 10.7a or 10.7b:

 Attempting stack trace sig 11
 Signal caught: signo [11]
 vsim_stacktrace.vstf written
 Current time Sun Sep 16 10:40:20 2018
 Program = vsim
 Id = "10.7b"
 Version = "2018.06"
 Date = "Jun  7 2018"
 Platform = linux_x86_64
 Signature = 86de93f3f985a4fb13b18b6839688b09
 --> START OF USERCODE
 0    0x00007ffff0ac36f4: 'PyEval_EvalCodeEx + 0x114' in '/pkg/qct/software/python/3.4.2/lib/libpython3.4m.so.1.0'
 1    0x00007ffff0a299bc: 'function_call + 0xbc' in '/pkg/qct/software/python/3.4.2/lib/libpython3.4m.so.1.0'
 2    0x00007ffff09fa5ad: 'PyObject_Call + 0x6d' in '/pkg/qct/software/python/3.4.2/lib/libpython3.4m.so.1.0'
 3    0x00007ffff0a14c3d: 'method_call + 0x9d' in '/pkg/qct/software/python/3.4.2/lib/libpython3.4m.so.1.0'
 4    0x00007ffff09fa5ad: 'PyObject_Call + 0x6d' in '/pkg/qct/software/python/3.4.2/lib/libpython3.4m.so.1.0'
 5    0x00007ffff0aba7a6: 'PyEval_CallObjectWithKeywords + 0x56' in '/pkg/qct/software/python/3.4.2/lib/libpython3.4m.so.1.0'
 6    0x00007ffff11a4271: 'gpi_log + 0x4dd' in '/prj/tbs/5g/sandiego/hw/main/users/hctseng/dev5g/txmdm/blocks/tx_modem/verif/sim/cocotb/build/libs/x86_64/libgpilog.so'
 7    0x00007ffff13bc77f: 'system_function_overload + 0x107' in '/prj/tbs/5g/sandiego/hw/main/users/hctseng/dev5g/txmdm/blocks/tx_modem/verif/sim/cocotb/build/libs/x86_64/libvpi.so'
 <-- END OF USERCODE

Please help!!

mciepluc, 2018-12-14:
I just tested on QuestaSim 10.7c, and everything is fine. Looks like accidental issue or bug in QuestaSim 10.7b.

ktbarrett, 2020-04-21:
I have run the regression with Questa 10.7b, I see no issue. This can be closed.

closed 2020-04-22 by imphil