cocotb/cocotb #667: Integration with test frameworks (pytest, unitetest...)

Nic30, 2018-10-19:
Hello,

is it possible to execute cocotb testbench directly from python without using a makefile.

(
I would like to pass large amount of configuration to test and makefile (or config file) is not a clean solution.

And I also have some hardware developement framework [hwt](https://github.com/Nic30/hwt) which has integrated simulator which I would like to replace it with cocotb.

Reason for this is that we have a large amount of simulation agents (drivers/monitors) for many interfaces and they are nearly cocotb compatible. (Not just AXI, Avalon, I2C but also ACE, PCI-e, DP, MIPI and others which are not commonly seen in opensource.) Our project is partially replica of yours and we would like to fix it.
)

If it is not currently possible I see that you probably has lots of other work to do. Are there any responsible persons for maintenance or any progressive fork?

tsengr0916, 2018-10-19:
Just curious. Do you use UVM version of verification IPs or do you have in-house build VIPs that are compatible with COCOTB? What is your integration approach to use your non-cocotb standard VIPs?

I don’t think there is any particular person responsible for anything in COCOTB. I filed a couple tickets before, no one has ever replied nor any comments. I kind of worked thru/around all problems myself.

Richard

Nic30, 2018-10-19:
@tsengr0916 We have UVM versions for some interfaces (XGMI, Axi-Lite). But I have many UVM like VIPs. More precisely, they are not in SystemVerilog but in python in framework similar to cocotb.

TC01, 2018-10-25:
> I would like to pass large amount of configuration to test and makefile (or config file) is not a clean solution.

Can you elaborate here? How would you like to pass the configuration data instead?

The cocotb makefile can be made much more complicated than the simple "define TOPLEVEL, TOPLEVEL_LANG, MODULE, and include the boilerplate makefiles". One thing I've done is create multiple simulation targets in my makefile; then switching between setups is as simple as running "make foo" vs "make bar". The different targets can then select different files to include, different toplevels, set environment variables to pass to the tests, etc.

If you really don't want to use make, I suppose you could try to replicate everything that happens in [Makefile.inc](https://github.com/potentialventures/cocotb/blob/master/makefiles/Makefile.inc) and [Makefile.sim](https://github.com/potentialventures/cocotb/blob/master/makefiles/Makefile.sim) by hand. But I've never tried to do it, and I suspect it would be pretty tricky.

Nic30, 2018-10-25:
One of my problem is:

(situation)
* I have configurable DMA engine which supports AXI3/4/Lite, ACE interfaces which are specified as a class and passed as an argument of constructor of python object ([generator](https://github.com/Nic30/hwt) of this DMA engine is written in python)
* In simulation I have a UVM-like interface agents which are connected into cached memory and a specific agent which controls registers of DMA engine itself.

(current state)
* Interface agents can be resolved from classes of interfaces (there is simple method on interface class which is returning the correct agent)
* Simulation uses multiple scenarios which runs in parallel in regular python test frameworks.

(the problem)
* Now user can simply configure python object of DMA and just run simulation as a python function.
(SImulation corutines from agents are automatically added, res has to be manually added by user.)
This allows to of python test frameworks which are reporting problems and doing parallelization.

If Makefile is used:
* Every configuration has to be serialized and deserialized by user
* Parallelization may require an extra effort
* It is not possible to use python HDL generator which has information about interfaces to configure simulation
* "It is not possible to generate HDL sources on demand"
* reporting of errors requires an extra effort
* Test automatization in python is complicated
* Python is executed many times, which takes time.

Nic30, 2018-10-25:
I expect that I help you to extract cocotb as a regular python module. I do not know when, but I hope it will be soon.

cmarqu, 2018-10-25:
A similar problem exists when you have an existing design and a complicated environment around it (lots of extra simulator switches, scripts being kicked before the compiler is even started etc.) and want to try cocotb on that design.
Instead of trying replicate that complicated environment in a cocotb Makefile, I would rather turn this around and add a handful of options and environment variables to the existing setup, so I fully support the idea behind this ticket.

nevsan, 2018-10-25:
Part of the problem is that the simulator is actually the topmost process and you would like Python to be the topmost process.

You can fake it by calling `make` using `subprocess` and communicating with it. I'm not saying it's the simplest way, but I've been able to get a nice link between the "main" Python and the one launched by the simulator to run `Cocotb` by using `pyzmq` to pass configuration data and results to and from using sockets. The end result is something that "looks" like a function call in Python, but really launches a simulation to consume arguments and produce a return value. The fact that you can so easily create a socket to send configuration data to your simulation is one of the beauties of having Python in your verification flow.

Nic30, 2018-10-25:
@nevsan I know that it can be emulated. But it is potentially infinite resource of issues.

I think that it is not possible to control simulator from python at least in some simulators.
I know that verilator simulations does support this. Do you know which simulators supports delta-stepping (can process only one simulation step on call)?

TC01, 2018-10-25:
Hi @Nic30,

> * It is not possible to use python HDL generator which has information about interfaces to configure simulation
> * "It is not possible to generate HDL sources on demand"

I guess I don't understand why these things are not possible. You can simply add another make target, which does something before the simulator even starts (as long as your target runs before the "sim" target). For instance, you could run some tool which generates HDL sources, puts them in the place referred to by ```VERILOG_SOURCES```, and then proceeds.

You could even generate some Python module using an existing script, which your tests then import, and use for configuration. (This generated code could even include cocotb tests, if you wanted).

e.g. do this:

```
my_target: do_something sim

do_something:
    # generate sources
    # do something to configure simulation
```

That being said, I certainly agree with @cmarqu that doing this isn't always ideal, and I imagine it can be painful to adapt an existing setup :)

One thing I've sometimes felt is missing from cocotb is some kind of "configure script" (or "cmake"?) layer which generates the Makefiles on demand, perhaps in response to switches on the command line. That might be a better place to do this sort of pre-simulation configuration? If nothing else it would help avoid extremely complicated makefiles.

I suspect that doing this sort of thing will be much simpler than trying to run cocotb from inside another Python process.

Nic30, 2018-10-26:
Hello @TC01,

 > * It is not possible to use python HDL generator which has information about interfaces to configure simulation
> * "It is not possible to generate HDL sources on demand"

I said it wrong. I meant that it is not possible without serialization/deserialization process.
I know that `pickle` can be used but still it is not working out of the box.

But back to my last question: Which simulators potentially supports python as topmost process?

closed 2020-11-24 by github-actions[bot]