cocotb/cocotb #668: Cocotb standard simulation agent (monitor/driver) library

Nic30, 2018-10-19:
Hello,

there are some gits with monitor/drivers for cocotb but it is sometimes hard to find and IO is not always consistent. It would be great to have global library of simulation monitors/driver for each interface.

Maybe it is not good to add extra work to you and community should create this library.

Let's search for libraries with simulation agents and build such a library.

For example:
@Dolu1990 has https://github.com/SpinalHDL/CocotbLib
@aoeldemann has https://github.com/aoeldemann/cocotb

cmarqu, 2018-10-25:
Roughly related: #220, talking about using existing verification IP.

themperek, 2018-11-21:
Yes.  We will move all the protocols out of cocotb to extension packages also coverage and verification methodology.  How exactly this will be done is to be decided some first examples should come soon @imphil

cmarqu, 2019-01-05:
Related: the second section of https://github.com/potentialventures/cocotb/wiki/Further-Resources

kolojoe, 2019-01-08:
Since this seems to have been answered should it be closed?

Nic30, 2019-01-08:
Ok, closing.

closed 2019-01-08 by Nic30